DexterInd/GrovePi #61: Add support for Grove Chainable RGB LED

mcauser, 2015-01-20:
http://www.seeedstudio.com/depot/Grove-Chainable-RGB-LED-p-850.html

mcauser, 2015-01-20:
http://www.dexterindustries.com/forum/?topic=support-for-grove-chainable-rgb-led

karan259, 2015-01-21:
Hey Mike,
Any update on this. Do you think that it would be better to cleanup and optimize the firmware source, i.e., go for 1.3 and then add the support for it or before that

mcauser, 2015-01-22:
See PR: https://github.com/DexterInd/GrovePi/pull/68

mcauser, 2015-01-23:
https://github.com/DexterInd/GrovePi/blob/master/Firmware/Source/v1.2/grove_pi_v1_2_2/README.md

BandonRandon, 2015-12-19:
Does  3be4d0c close this?

karan259, 2015-12-19:
Yeah, the grove RGB led's now work. There is another issue with connecting multiple sensors. You can read more about it here: https://github.com/DexterInd/GrovePi/issues/180

closed 2015-12-19 by karan259

mcauser, 2015-12-20:
I have 10x Grove RGB LEDs. I'll do some debugging. It's a 4 wire module, (data, clock, vcc, gnd) so if you are using D4, you must keep D5 empty as it shares a pin.