DexterInd/GrovePi #83: Simple robot project with Java support for some Grove modules

jbman, 2015-03-08:
Hi, 
I've built a simple driving robot called "Captain Picar" with the GrovePi and Java.
Here is the source: https://github.com/jbman/captain-picar

Anyone who is interested in using  the GrovePi with Java might take it as an example. You may also add the generic code to the Software section of the GrovePi project (package com.github.jbman.jgrove). 

The project amkes use of the [pi4j I/O API and libraries][PI4J] project. 
Currently there are classes for these Grove modules: 

* Button  
* Ultrasonic
* MotorDriver 
* Lcd with RGB backlight
* (Digital Gyro started, but it produces to many errors when reading the values)

[PI4J]: http://pi4j.com/

noelportugal, 2015-04-12:
Thanks @jbman  this is awesome! Do you know of any other Java library for GrovePi? If not you should create a separate project just for the library. Just a thought. Good job!

marcellobarile, 2015-04-12:
Hi guys, I've already made a Java library for the GoPiGo (you can find it in the GoPiGo repo) and I was planning to do the same for the GrovePi. I'll take a look to the @jbman 's project and I'll keep you in the loop. Enjoy!

closed 2016-04-22 by DexterInd