DexterInd/GrovePi #84: Busy loop vs Interrupts?

martinschaef, 2015-03-19:
I'm trying to build a clap sensor using the sound sensor module that switches lights on/off on double-clapping. However, due to the sleep in the analogRead, I miss claps from time to time. Increasing the sampling rate will result in a busy loop which is not good either. Is there a way that I can register to hardware timers / interrupts?

marcellobarile, 2015-03-20:
Hello Martin, which library are you using for your project?

martinschaef, 2015-03-20:
I'm using the python code that you have in the git:
https://github.com/DexterInd/GrovePi/blob/master/Software/Python/grovepi.py
and
https://github.com/DexterInd/GrovePi/blob/master/Software/Python/grove_sound_sensor.py

My stuff is here https://github.com/martinschaef/grovepi

marcellobarile, 2015-03-20:
so, basically, do you miss all those claps which occurs during the "sleep" (100ms), right?

martinschaef, 2015-03-20:
Yes ... I'd assume that the sleep in the analogRead is too long. However, I'm not sure if I can or want to make that shorter because than I'd end up in a busy loop which I don't want to run in background (However, I'm not sure how bad that is in python). Some interrupt driven thingy would be a lot nicer.

marcellobarile, 2015-03-20:
okay, let me give it some thought :)

martinschaef, 2015-03-20:
Great, Thanks. Your firmware is not open, right?

karan259, 2015-03-20:
The source for the GrovePi firmware is here: https://github.com/DexterInd/GrovePi/blob/master/Firmware/Source/v1.2/grove_pi_v1_2_2/grove_pi_v1_2_2.ino .

The biggest factor might be the delay that we added in the GrovePi.py code here: https://github.com/martinschaef/grovepi/blob/master/grovepi.py#L163. You can try removing that and see if that helps. It was there so that the receive buffer always have the latest values. (Removing it might also lead to errors too).

I2C has some overhead too, so I don't think you would be able to get sub millisecond updates.

martinschaef, 2015-03-21:
Tested it with the newest firmware and a smaller sleep in grovepi.py and it seems to work ok (still not perfect). Will continue playing with it.

closed 2015-03-21 by martinschaef

marcellobarile, 2015-03-22:
okay Martin, enjoy!