DexterInd/GrovePi #86: Installation Failure

CubicleJockey, 2015-03-23:
Below are the error sections I receive when trying to install node-grovepi via npm:

gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1067:12)
gyp ERR! System Linux 3.18.7-v7+
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node_modules/node-grovepi/node_modules/i2c
gyp ERR! node -v v0.12.0
gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok
npm ERR! Linux 3.18.7-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "node-grovepi"
npm ERR! node v0.12.0
npm ERR! npm  v2.5.1
npm ERR! code ELIFECYCLE

npm ERR! i2c@0.1.6 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the i2c@0.1.6 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the i2c package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls i2c
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/GrovePi/Practice/npm-debug.log

marcellobarile, 2015-03-23:
Hello CubicleJockey,
which version of "npm" and "node" do you have?
could you please post the content of "/home/pi/GrovePi/Practice/npm-debug.log" on http://pastebin.com/ and also the command that you are using to install the module? 

Thank you

CubicleJockey, 2015-03-24:
Hey. Thanks for getting back so quickly.

Sorry I am using a gist from github, since I didn't want to make any new accounts today.

But here are my repro steps: https://gist.github.com/CubicleJockey/3149fbd0db260b266b5b

Thank you for your time and effort.

marcellobarile, 2015-03-24:
Okay, got it. I'll be back to you as soon as I understand what's happened on your side :-) stay tuned.

marcellobarile, 2015-03-24:
I was looking through your logs and what seems to be strange is that you are trying to install the module inside your user's folder ("/pi/node_modules"). Could you try to install the module inside your application's folder OR globally thanks to the "-g" argument? (es: $ npm install -g node-grovepi)

BTW, the error is related  to the i2c module, which is a dependency of our module.

I look forward to your reply.

marcellobarile, 2015-03-24:
Okay, I probably got it... looking here https://github.com/kelly/node-i2c/issues/55 seems like we need to update the dependency to fully support the latest Node.js version.

Until we patch it, I suggest you to install i2c separately and then proceed with the installation.

Let me know if this helped you.

CubicleJockey, 2015-03-24:
Hi,

I see a lot of discovery happened before I could get back here today. Thanks for the effort.

Raspberry Pi Model: Pi 2 Model B

I also have a Model B Revision 2.0 Mounting holes, I could load up a fresh Raspian and try there as well.

marcellobarile, 2015-03-24:
as far as I read, it seems that the 0.12.0 version is still only available for the Pi 2 Model (from the official repository) and our package.json contains an older version of the i2c module, which is not compatible with that version of Node.js; that's why you are facing troubles.

As said before, you can try to install the latest version of the i2c module separately and then continue to install the node-grovepi module. If necessary, you can also change your local package.json in order to download the latest version of the i2c module.

Let me know.

Best regards

CubicleJockey, 2015-03-25:
@marcellobarile alright so I haven't tested running it yet but this got me to build without error. I am being explicit with steps for others who may find this.

1. sudo npm install GrovePi
2. sudo npm install i2c -g (currenly version 2.1) 
3. cd GrovePi/Software/NodeJs/
4. sudo nano package.json
5.  "dependencies": {
           "i2c": "0.1.6",  <---- change to version 0.2.1
6. ctrl + x, and Y to save
7. sudo npm install -g
8. After installation completes you will see the library listed as:
     node-grovepi@1.0.2 /usr/local/lib/node_modules/node-grovepi
     ├── fs@0.0.2
     ├── async@0.9.0
     ├── npmlog@1.1.0 (ansi@0.3.0, are-we-there-yet@1.0.3, gauge@1.1.0)
     ├── buffertools@2.1.2
     ├── sleep@1.2.0 (mkdirp@0.3.5)
     └── i2c@0.2.1 (bindings@1.2.1, repl@0.1.3, underscore@1.8.2, coffee-script@1.9.1)

CubicleJockey, 2015-03-25:
@marcellobarile after that build I tried to run this script in a folder outside of the _GrovePi/Software/NodeJS_ folder and in it but it says it can't find 'node-grovepi'. As you see from my previous post I did a *sudo npm install -g* to try and make it global.

```javascript
(function(){
	'use strict';
	
	var grovepi = require('node-grovepi');
	
	grovepi.init({
		onInit: function callback(){
			grovepi.run([version(callback), light(callback)]);
		},
		onError: function(error){
			console.log(error);
		}
	});
	
	function version(callback){
		grovepi.version(function onVersion(response){
			console.log('Version'.concat(response));
			callback();
		})
	};
	
	function light(callback){
		var digitalPort = 4;
		grovepi.light(digitalPort, function onLight(response){
			console.log('Light'.concat(response));
			callback();
		});
	};
})()
```

After I get this error:
>module.js:338
>    throw err;
>          ^
>Error: Cannot find module 'node-grovepi'
>    at Function.Module._resolveFilename (module.js:336:15)
>    at Function.Module._load (module.js:278:25)
>    at Module.require (module.js:365:17)
>    at require (module.js:384:17)
>    at /home/pi/grovepi-starterkit-examples/NodeJS/blinkyLed.js:4:16
>    at Object.<anonymous> (/home/pi/grovepi-starterkit-examples/NodeJS/blinkyLed.js:29:3)
>    at Module._compile (module.js:460:26)
>    at Object.Module._extensions..js (module.js:478:10)
>    at Module.load (module.js:355:32)
>    at Function.Module._load (module.js:310:12)

I went to double check what npm modules are installed so I ran *npm ls*.

Result is:
>/home/pi
>├── UNMET DEPENDENCY npm-debug.log@*
>└── pi-gpio@0.0.6
>
>npm ERR! missing: npm-debug.log@*, required by undefined@undefined

I am guessing that work around isn't quite able to install properly even though it didn't complete after changing the version to 2.1.

marcellobarile, 2015-03-25:
Hi @CubicleJockey, could you try to perform a "npm list" inside your Node.js app's folder?

marcellobarile, 2015-04-08:
Hello @CubicleJockey, how is going with your issue? Did you find a way out?

CubicleJockey, 2015-04-08:
Hey sorry I got super distracted. My company shipped me over to take over a
coworkers spot who quit suddenly at Microsoft so I have been swamped. No I
haven't quite figured anything out yet but I haven't had time either. I'll
let you know if I do. Thanks.

marcellobarile, 2015-04-08:
okay, as last chance, I suggest you to install a previous version of Node (v0.10.25 works fine with the I2C dependant module). You can find all the necessary procedure here: http://raspberryalphaomega.org.uk/2014/06/11/installing-and-using-node-js-on-raspberry-pi/

I'll update the issue as soon as I got better news regarding the 0.12 Node.js version.

Enjoy!

CubicleJockey, 2015-04-08:
Awesome. Thank you for your support. Top Notch!

marcellobarile, 2015-06-24:
Hi @CubicleJockey , there is a new version of the node-grovepi module and you can find more info here: https://www.npmjs.com/package/node-grovepi
Please, note that it has been refactored and it will break your application if you switch from the v1.* to the v2.* ... on the other hand, it fully supports the latest version of Node.js

Enjoy!

CubicleJockey, 2015-06-24:
Thank you for the update. I am excited that the latest version is
available. But now I am conflicted betwee C# with the Win10 IOT on
Raspberry Pi 2 or try and work it out with NodeJS. Ahhh Choices. I may just
have to get multiple memory cards and build anything I build in both. :)

Cheers and thanks for the update.

closed 2015-08-26 by DexterInd