DexterInd/GrovePi #90: Provide an example of reading data from analog or digital port with i2cget in Shell examples

in8finity, 2015-04-03:
There are two examples. One is for display and one for fm radio. It would be very helpful to have a sample of data reading from port on groove board itself.

karan259, 2015-04-03:
The analog or digital read are multistep process. Do you think that making a python script executable and having it return just the value back to the shell would work.

in8finity, 2015-04-03:
I understand but in my case python is excessive dependency. Probably I should try to compile C versions. It will be lighter and faster.
I'm I right that it will be process like described below?
1) i2cset -y 1 0x04 0x03 0x01 0x00 i; //analogRead to Groove
2) i2cset -y 1 0x04 0x05 0x01 0x00 i; //set pin into input mode 
...?
3) i2cget -y 1 0x04; // but it's getting zero

i2cdump returns data containing the right answer (and then Groove hangs up)

mcauser, 2015-05-13:
In your step 3) you need to read block data, not just a single byte.

```
# set pin mode to input on port A0
# send block [5 0 0 0]
i2cset -y 1 0x04 1 5 0 0 0 i

# perform an analog read on port A0
# send block [3 0 0 0]
i2cset -y 1 0x04 1 3 0 0 0 i

# 100 millisecond pause
# give the grovepi some time to process the command
sleep .1

# read a byte
i2cget -y 1 0x04 0

# read i2c block data
i2cget -y 1 0x04 0 4
```

closed 2016-04-25 by DexterInd