DexterInd/GrovePi #105: Running test programs using GrovePi+ board from idle Python 2 or 3

oliwinkler, 2015-05-08:
I can run without problem athe test program grove_relay.py from the command line : sudo python grove_relay.py

When I want to run it from idle python I get an error message when doing an import grovepi

bus = smbus-SMBus(1)
IOError: [Erno 13] Permission denied. .......

Questions
a) How to run from the command line python grove_relay.py not as SUDO?
or
b) from Idle,  python grove_relay.py  as SUDO when Running the program?

How to set up the corresponding right?

Thank you

Olivier

kickapoo, 2015-09-22:
You need to be root user to access GPIO. 

Try to set grant root access:
`sudo usermod -a -G sudo <username>`

You need to log off and re-enter in order to take affect.

karan259, 2015-09-22:
To grant access to the I2C to user pi, use:
```sudo adduser pi i2c```

closed 2016-04-22 by DexterInd