DexterInd/GrovePi #130: please update package.son for node library!

mrose17, 2015-08-04:
npm sleep version 2.0.* no longer compiles on the RPi ... you need to be using 3.0.* in https://github.com/DexterInd/GrovePi/blob/master/Software/NodeJS/libs/package.json

marcellobarile, 2015-08-04:
Hi @mrose17,
I'm doing some tests before to bump the version of the sleep module dependency and after that of course it will be updated. Thank you for your cooperation :-)

mrose17, 2015-08-04:
thanks! i haven't studied the package to understand why you're using sleep. i suspect that an async read might be more "node-like". if i have some ideas, i'll let you know.

marcellobarile, 2015-08-04:
Sure, any help will be appreciated :-) Let's keep in touch.

mrose17, 2015-08-06:
hi. just a few notes:

1. i've been using https://github.com/mrose17/GrovePi -- which is identical to your upstream repo -- except for the one change to package.json, without any issues on my RPi1 B+ and RPi 2, without any issues.

2. in doing some testing, i've found three sensors that don't seem to have support. are these "known" limitations?

* [Grove - Barometer Sensor](http://www.seeedstudio.com/wiki/Grove_-_Barometer_Sensor - I2C)

* [Grove - Digital Light Sensor](http://www.seeedstudio.com/wiki/Grove_-_Digital_Light_Sensor)

* [Grove - Dust Sensor](http://www.seeedstudio.com/wiki/Grove_-_Dust_Sensor)

finally, thanks very much for doing the Node.js port ... it is very helpful! i hope to point you to a repo i've just written that uses your port! (probably in a day or so)

i have two substantive comments:

first, i'd put the node package in it's own repository, and use `git submodule` to link it to `Software/NodeJS/libs`

second, use of `sleep` is _contra-Node.js_. any function that calls sleep, would be better off by adding a final `callback` parameter, e.g.,

        GrovePi.prototype.initialize = function(cb) {
          if (!isHalt) {
            bus = i2c.openSync(busNumber)
        
            if (!isInit) {
              this.debug('GrovePi is initing')
              setTimeout(function () {
                  isInit = true
                  cb(null, true)
              }, initWait * 1000)
            } else {
              cb(null, false)
            }
          } else {
            cb(new Error('GrovePI cannot be initialized'))
          }
        }

        GrovePi.prototype.readVersion = function(cb) {
          var write = this.writeBytes(commands.version.concat([commands.unused, commands.unused, commands.unused]))
          if (!write) return false

          setTimeout(Function () {
            this.readByte()
            var bytes = this.readBytes()

            if (typeof bytes == 'object')
              cb (null, bytes[1] + '.' + bytes[2] + '.' + bytes[3])
            else
              cb(null, false)
          }, 100)
          return true
        }

does this make sense? note that i have changed the prototype names so as not to conflict with the existing API.

would you like me to submit a PR with changes like this?

thanks!

/mtr

marcellobarile, 2015-08-06:
Hey @mrose17 thank you for your contribute! I've just published the bumped version of the sleep module dependency! Regarding the "sleep" vs "timeout" thing, I use the sleep module because a complete freeze of the code execution is needed in order to let the board do its own job. I started using timeouts and callbacks but sometimes the reads were pretty unstable, that's why I switched to the sleep module.

About the dedicated repository, I need to involve the other guys so it's probably better to create another issue only for this. 

Regarding the missing sensors, please, feel free to write your own classes and open a pull request; I'll really appreciate it :-)

Thank you and let me know if you have any other doubt!

mrose17, 2015-08-06:
thanks for the update!

i'm surprised to hear that using a timeout would cause a problem with the reads. whether you are using sleep or setTimeout, the kernel runs independently in terms of getting input from pins. it should not matter, it really shouldn't... perhaps i'll run an experiment this weekend.

when you write _pretty unstable_, what were you seeing?

also, i'll see if i can find some time to write some new sensor modules.

many thanks!

closed 2015-08-06 by mrose17

mrose17, 2015-08-10:
@marcellobarile - just an update for you: https://github.com/mrose17/homespun -- i'll summarize it: it's a node.js script that reads a config file telling it what sensors are where on the GrovePi, reads them and transcodes them to TSRP (you'll find a link on the README). whatever is listening for TSRP traffic can then upload the statistics to the cloud. more work coming up...

marcellobarile, 2015-08-10:
nice one! Thank you!