DexterInd/GrovePi #143: cloning from raspberry pi

brat-ibr, 2015-09-02:
Hi Guys,
Doubt if this is a NEW issue but....
Trying to setup my Raspberry pi for grovepi and get message - Couldn't resolve host 'gitup.com' while accessing ...../GrovePi.git/info/refs
Thank you

karan259, 2015-09-02:
We haven't seen this issue before but it looks like either there is a spelling mistake when you are cloning the repo. it must be **github** and not **gitup**. Can you try git clone on the Pi: ```https://github.com/DexterInd/GrovePi.git``` to clone the repo and make sure that internet works on the Pi. An easy way to test this would be to ping google: ``ping google.com````. This guide might help you setting everything up: http://www.dexterindustries.com/GrovePi/get-started-with-the-grovepi/setting-software/ .

brat-ibr, 2015-09-03:
Hi Guys,
Thanks for your swift reply. I know that it is me being stupid but that 
goes with the territory :-)) sorry.

Your details on
http://www.dexterindustries.com/GrovePi/get-started-with-the-grovepi/setting-software/
show the second ecreen with the prompt line as 
root@raspberrypi:/home/pi/DEsktop# git clone ... etc
SO
Using Putty logged in via wireless to my RPI ;  used su - to become sudo 
user ; and attached file show my 'adventures' .
Please tell me what it is I have missed ???

many thanks,  ianm
PS I am NOT commercial. Just me.

-- 
Impala Bipedal Robotics
ian mitchell
+61414877990
alt email: ianmitchell777@optusnet.com.au

DexterInd, 2015-09-03:
Hey Ian, I don't see anything attached here.  Did we miss something?

Glad to see more about your adventures, this is great!

brat-ibr, 2015-09-09:
Solved thx guys.
8.8.8.8 did not test connection for DNS which WAS the problem.. All good 
now thank you.

ianm

-- 
Impala Bipedal Robotics
ian mitchell
+61414877990
alt email: ianmitchell777@optusnet.com.au

closed 2015-11-20 by karan259