DexterInd/GrovePi #144: Grove LCD and NodeJS

k33g, 2015-09-14:
Hello,

I’m a newbie with GrovePi and I’ve understood how to extend AnalogicSensor to use sound sensor. But it becomes more complicated (for me) with Grove LCD and Leds. It would be fine to have some samples about LCD & leds.

Thank you

Regards
Philippe

marcellobarile, 2015-09-17:
Hello @k33g,
I'll try to provide you something useful for your needs. Let's keep in touch, okay?

k33g, 2015-09-17:
oh nice :smiley:  :+1:  thank you very much :smile:

danibram, 2016-08-30:
Can you post here? I needed too

beckybettinger, 2016-12-03:
Hey, i'm also curious :) Is there any example code on how to use the 96x96 OLED Display with node?

jlorek, 2016-12-20:
Im also looking for an example or at least an direction on how to use the display in NodeJS.
Any help is appreciated!
Thanks :)

marcellobarile, 2016-12-22:
Sorry for being late guys, I don't have that display but I'll do my best to give you some reasonable answer :)

danibram, 2016-12-22:
Hey guys,
I wrote this class for a workshop with my groove pi
`
const GrovePi = require('node-grovepi').GrovePi;
const fs = require('fs');
const sleep = require('sleep');
const extend = require('extend');

const I2cSensor = GrovePi.sensors.base.I2C;
const i2cBus = require('i2c-bus');

const DISPLAY_RGB_ADDR = 0x62;
const DISPLAY_TEXT_ADDR = 0x3e;

const i2c0Path  = '/dev/i2c-0'
const i2c1Path  = '/dev/i2c-1'
const i2c2Path  = '/dev/i2c-2'

class LCD_I2C {
    constructor(){
        this.data = {}

        this.initialized = false;
        this.bus = null;
        // Bus Address from i2cdetect command
        this.address = 0x68;
        // Default delay for streaming
        this.watchDelay = 100;

        let busNumber

        if (fs.existsSync(i2c0Path)) {
          busNumber = 0
        } else if (fs.existsSync(i2c1Path)) {
          busNumber = 1
        } else if (fs.existsSync(i2c2Path)) {
          busNumber = 2
        } else {
          var err = console.log('ERR: Could not determine your i2c device')
        }

        this.i2c1 = i2cBus.openSync(1);
        console.log(this.i2c1)

        this.setText('Ready!');
        this.setRGB(55, 55, 255);
    }

    //data
    setData(data) {
        this.data = Object.assign({}, this.data, data)
        this.update(this.data)
    }

    update(data) {
        var text = ""
        Object.keys(data).map((k) => {
            text = text + ` ${k}: ${data[k]}`
        })
        this.setText(text);
    }

    //Hardware
    setRGB(r, g, b) {
      this.i2c1.writeByteSync(DISPLAY_RGB_ADDR,0,0)
      this.i2c1.writeByteSync(DISPLAY_RGB_ADDR,1,0)
      this.i2c1.writeByteSync(DISPLAY_RGB_ADDR,0x08,0xaa)
      this.i2c1.writeByteSync(DISPLAY_RGB_ADDR,4,r)
      this.i2c1.writeByteSync(DISPLAY_RGB_ADDR,3,g)
      this.i2c1.writeByteSync(DISPLAY_RGB_ADDR,2,b)
    }

    textCommand(cmd) {
      this.i2c1.writeByteSync(DISPLAY_TEXT_ADDR, 0x80, cmd);
    }

    setText(text) {
      this.textCommand(0x01) // clear display
      sleep.usleep(50000);
      this.textCommand(0x08 | 0x04) // display on, no cursor
      this.textCommand(0x28) // 2 lines
      sleep.usleep(50000);

      var count = 0;
      var row = 0;
      for(var i = 0, len = text.length; i < len; i++) {
        if(text[i] === '\n' || count === 16) {
          count = 0;
          row ++;
            if(row === 2)
              break;
          this.textCommand(0xc0)
          if(text[i] === '\n')
            continue;
        }
        count++;
        this.i2c1.writeByteSync(DISPLAY_TEXT_ADDR, 0x40, text[i].charCodeAt(0));
      }
    }
}

export default LCD_I2C
`

I hope this help you, its written in es6 but you can easy extract the basic idea

marcellobarile, 2016-12-22:
hello @danibram,
that's awesome, thank you for your contribution!

closed 2017-12-07 by johnisanerd