DexterInd/GrovePi #172: Not able to install npm package on Visual Studio

rhtnryn, 2015-11-26:
====Executing command 'npm install node-grovepi --save '====
npm WARN package.json DemoNodeApp@0.0.0 No repository field.
npm WARN package.json DemoNodeApp@0.0.0 No README data
npm WARN package.json DemoNodeApp@0.0.0 No license field.
> buffertools@2.1.3 install C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools
> node-gyp rebuild
C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools>if not defined npm_config_node_gyp (node "C:\Program Files (x86)\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node  rebuild ) 
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  buffertools.cc
c:\users\rohit narayan\documents\visual studio 2015\projects\demonodeapp\demonodeapp\node_modules\node-grovepi\node_modules\buffertools\BoyerMoore.h(35): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
c:\users\rohit narayan\documents\visual studio 2015\projects\demonodeapp\demonodeapp\node_modules\node-grovepi\node_modules\buffertools\BoyerMoore.h(57): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
c:\users\rohit narayan\documents\visual studio 2015\projects\demonodeapp\demonodeapp\node_modules\node-grovepi\node_modules\buffertools\BoyerMoore.h(61): warning C4267: 'initializing': conversion from 'size_t' to 'int', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
c:\users\rohit narayan\documents\visual studio 2015\projects\demonodeapp\demonodeapp\node_modules\node-grovepi\node_modules\buffertools\BoyerMoore.h(61): warning C4267: 'initializing': conversion from 'size_t' to 'const int', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
c:\users\rohit narayan\documents\visual studio 2015\projects\demonodeapp\demonodeapp\node_modules\node-grovepi\node_modules\buffertools\BoyerMoore.h(62): warning C4267: 'initializing': conversion from 'size_t' to 'int', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
c:\users\rohit narayan\documents\visual studio 2015\projects\demonodeapp\demonodeapp\node_modules\node-grovepi\node_modules\buffertools\BoyerMoore.h(62): warning C4267: 'initializing': conversion from 'size_t' to 'const int', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
c:\users\rohit narayan\documents\visual studio 2015\projects\demonodeapp\demonodeapp\node_modules\node-grovepi\node_modules\buffertools\BoyerMoore.h(110): warning C4267: '=': conversion from 'size_t' to 'int', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
..\buffertools.cc(219): warning C4267: 'initializing': conversion from 'size_t' to 'int', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
..\buffertools.cc(219): warning C4267: 'initializing': conversion from 'size_t' to 'const int', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
..\buffertools.cc(220): warning C4267: 'initializing': conversion from 'size_t' to 'int', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
..\buffertools.cc(220): warning C4267: 'initializing': conversion from 'size_t' to 'const int', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
..\buffertools.cc(332): warning C4267: '=': conversion from 'size_t' to 'int32_t', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
..\buffertools.cc(334): warning C4267: '=': conversion from 'size_t' to 'int32_t', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
..\buffertools.cc(340): warning C4244: 'argument': conversion from 'const ptrdiff_t' to 'int32_t', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
..\buffertools.cc(419): warning C4267: 'argument': conversion from 'size_t' to 'int', possible loss of data [C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\buffertools.vcxproj]
  win_delay_load_hook.c
     Creating library C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\Release\buffertools.lib and object C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\Release\buffertools.exp
  Generating code
  Finished generating code
  buffertools.vcxproj -> C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\buffertools\build\Release\\buffertools.node
> sleep@3.0.0 install C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\sleep
> node-gyp rebuild
C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\sleep>if not defined npm_config_node_gyp (node "C:\Program Files (x86)\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node  rebuild ) 
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  sleep.cc
  win_delay_load_hook.c
     Creating library C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\sleep\build\Release\node_sleep.lib and object C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\sleep\build\Release\node_sleep.exp
  Generating code
  Finished generating code
  node_sleep.vcxproj -> C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\sleep\build\Release\\node_sleep.node
> i2c-bus@0.11.3 install C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\i2c-bus
> node-gyp rebuild
C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\node_modules\node-grovepi\node_modules\i2c-bus>if not defined npm_config_node_gyp (node "C:\Program Files (x86)\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node  rebuild ) 
Building the projects in this solution one at a time. To enable parallel build, please add the "/m" switch.
  i2c.cc
[... cut ...]
npm ERR! This is most likely a problem with the i2c-bus package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls i2c-bus
npm ERR! There is likely additional logging output above.
npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\Rohit Narayan\documents\visual studio 2015\Projects\DemoNodeApp\DemoNodeApp\npm-debug.log
====npm command completed with exit code 1====

marcellobarile, 2015-11-26:
Hi @rhtnryn,
could you please tell me more about the environment of your project?

rhtnryn, 2015-11-26:
This is a universal windows platform application project based on Node.js for Windows IoT core and I'm using Visual Studio 2015 IDE.

marcellobarile, 2015-11-26:
Are you running it on a Raspberry Pi?

rhtnryn, 2015-11-26:
Yes, the project has to be deployed on Raspberry Pi.

marcellobarile, 2015-11-26:
To be honest I've never wrote applications for the RPi with the Visual Studio so I don't really know the workflow... I'm just trying to understand if you are getting this error on a RPi or on a Personal Computer. From your log I can clearly see that there are some troubles during the compiling of the I2C library:

i2c.cc
c:\users\rohit narayan\documents\visual studio 2015\projects\demonodeapp\demonodeapp\node_modules\node-grovepi\node_modules\i2c-bus\src./i2c-dev.h(26): fatal error C1083: Cannot open include file: 'linux/types.h': No such file or directory [C:\Users\Rohit Narayan\documents\visual studio

rhtnryn, 2015-11-26:
i'm getting this error on my personal computer

marcellobarile, 2015-11-26:
You can develop your Node application whenever you want but, after that, you should run it on your RPi (because you are going to use some hardware which is absent on your Personal Computer). I don't know your skill level on Node.js but you can easily install the application through  the "npm install" command after having created a package.json file (with dependencies etc...)

Let me know if you need further informations.

Have a good day

closed 2017-12-07 by johnisanerd