DexterInd/GrovePi #175: Cannot build VS2015 project with grovepi nuget reference

ser1us, 2015-12-09:
Hello!

Installed the GrovePi nuget package into my c# windows iot backgroud task template. But I always get the following error:

```
APPX0702: Payload file [USER-Folder]\.nuget\packages\GrovePi\1.0.6\lib\uap10.0\GrovePi\Properties\client.winU.rd.xml' does not exist.
```

Anyone knows what the problem could be?

Best regards,
Christoph

penfold, 2015-12-29:
Having the same issue. Any further info available?

chumiporky, 2016-01-02:
I have the same issues, I will comment on the repository of the creator of the nugget package, robsonj

dougseven, 2016-01-31:
Trying to blink an LED and get this when deploying to the Pi: 

`Payload file 'C:\Users\<username>\.nuget\packages\GrovePi\1.0.6\lib\uap10.0\GrovePi\Properties\client.winU.rd.xml' does not exist.`

Has there been any resolution to this?

maxpg, 2016-01-31:
Same for me. I downloaded the project and added it directly as a project to my solution.

lanselambor, 2016-01-31:
Copy 
[ GrovePi/Software/CSharp/GrovePi/Properties/ ]
to 
[ C:\Users\<Your computer name>\.nuget\packages\GrovePi\1.0.6\lib\uap10.0 ]

solved the problem.

danvy, 2016-03-09:
Issue is still opened. Could you update the NuGet package?

rpedersen, 2016-04-15:
I have fixed this issue with PR #228.

DexterInd, 2016-04-17:
Just accepted @rpedersen pull request on this issue.

closed 2016-04-22 by DexterInd