DexterInd/GrovePi #181: Could not locate the bindings file.

AdamMiltonBarker, 2015-12-27:
Error: Could not locate the bindings file. Tried:
 → /data/jumpway/iot/node_modules/i2c-bus/build/i2c.node
 → /data/jumpway/iot/node_modules/i2c-bus/build/Debug/i2c.node
 → /data/jumpway/iot/node_modules/i2c-bus/build/Release/i2c.node
 → /data/jumpway/iot/node_modules/i2c-bus/out/Debug/i2c.node
 → /data/jumpway/iot/node_modules/i2c-bus/Debug/i2c.node
 → /data/jumpway/iot/node_modules/i2c-bus/out/Release/i2c.node
 → /data/jumpway/iot/node_modules/i2c-bus/Release/i2c.node
 → /data/jumpway/iot/node_modules/i2c-bus/build/default/i2c.node
 → /data/jumpway/iot/node_modules/i2c-bus/compiled/5.3.0/linux/arm/i2c.node
    at bindings (/data/jumpway/iot/node_modules/bindings/bindings.js:88:9)
    at Object.<anonymous> (/data/jumpway/iot/node_modules/i2c-bus/i2c-bus.js:4:28)
    at Module._compile (module.js:398:26)
    at Object.Module._extensions..js (module.js:405:10)
    at Module.load (module.js:344:32)
    at Function.Module._load (module.js:301:12)
    at Module.require (module.js:354:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (/data/jumpway/iot/node_modules/node-grovepi/grovepi.js:2:19)
    at Module._compile (module.js:398:26)
    at Object.Module._extensions..js (module.js:405:10)
    at Module.load (module.js:344:32)
    at Function.Module._load (module.js:301:12)
    at Module.require (module.js:354:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (/data/jumpway/iot/node_modules/node-grovepi/index.js:3:12)

marcellobarile, 2015-12-29:
Hi @AdamMiltonBarker
could you tell me more about your environment? What's your OS, version of Node.js etc... thank you.

DexterInd, 2016-04-17:
Hi @AdamMiltonBarker any progress or response for @marcellobarile ?

closed 2016-04-25 by DexterInd