DexterInd/GrovePi #206: grove collision sensor

csoukup, 2016-02-26:
I am unable to get the grove pi collision sensor to detect a collision using grove_collision_sensor.py.  Please help.

DexterInd, 2016-02-26:
What's the issue?

dsoukup, 2016-02-29:
I am using the grove_collision sensor:  http://www.seeedstudio.com/depot/Grove-Collision-Sensor-p-1132.html
I do notice that it appears to be slightly different than the one shown on the wiki page:
http://www.seeedstudio.com/wiki/Grove_-_Collision_Sensor

When I run the program the grove_collision_sensor.py program, the output is 0, and it seems inconsistent with what is happening to the sensor.  I did see a reading of 255 one time, but again, it didn't seem to correlate with what is happening to the sensor.

The information that I find on it says that it goes low when it detects vibration, but the information says that normal shaking does not cause any output.  I am not experiencing this to be true.

Please advise.  Thank you!

dsoukup, 2016-02-29:
My apologizes.  I meant to say that I am getting an output of 1.  I can get outputs of 1 and 255, but, as I was saying, the output is not consistent with what is happening to the sensor.

DexterInd, 2016-02-29:
Have you tried it out on a regular Grove system?  I'm not totally familiar with the code, but my best guess is that if you're not getting a response, you might have received some bad hardware?

dsoukup, 2016-02-29:
I wish it were just bad hardware, but I have 4 (one from a different
vendor) and I have swapped out the collision sensors already to check if it
was just a faulty hardware issue.  However, the issue persists with all 4
collision sensors.

Today, I also noticed that I am having an issue with the 3 axis digital
accelerometer (+-16), which reads zero on the tilt buzzer program.  When I
tested it with the example program in the values read very low decimals,
near zero.

If I understand correctly, the collision sensor is an accelerometer too --
so I am wondering if something larger is creating a problem with
accelerometer readings for me.

The tilt buzzer program is also acting up with the switch, reading the
switch is off at times when it is in the high position (and then returning
to reading the accelerometer, albeit what appears to be erroneous values of
low decimals close to zero) and then returning to reading as off (without
any intervention by the user).

I did recently update the firmware.

I very much appreciate help with these issues because I am a teacher, and
my students are participating in a city-wide IoT challenge.  Getting these
sensors properly working is essential to two teams of students' working
prototypes which they are demonstrating on Thursday for city tech leaders.
Thank you so much!

karan259, 2016-02-29:
Hey @dsoukup,
It would be great if we could continue the discussion on the Forums here: http://www.dexterindustries.com/forum/grovepi/. We don;t have that sensor and it was a user contributed code, but from the looks of the Seeed code, it should work. Can you post some sample output of what you are getting. Are you not getting an output 0 on it. Can you also post a few pictures of your setup on the forums

closed 2016-03-23 by DexterInd