DexterInd/GrovePi #214: win10 Csharp SimpleDriver.cs occur Exception

lanselambor, 2016-03-11:
This is the steps I run the code:
1)Download GrovePi and click **GrovePi\Software\CSharp\GrovePi.sln** to open the project with VisualStudio 2015.
2)Seeting SimpleDriver.cs as **Startup Project**,
3)Compile and deploy the project.
4)Exception happended after a few minutes.

<pre>
'backgroundTaskHost.exe' (CoreCLR: DefaultDomain): Loaded 'c:\data\Programs\WindowsApps\Microsoft.NET.CoreRuntime.1.0_1.0.23430.0_arm__8wekyb3d8bbwe\mscorlib.ni.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'backgroundTaskHost.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Data\Users\DefaultAccount\AppxLayouts\GrovePiVS.Debug_ARM.lambor\Driver.winmd'. Symbols loaded.
'backgroundTaskHost.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Data\Users\DefaultAccount\AppxLayouts\GrovePiVS.Debug_ARM.lambor\System.Runtime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'backgroundTaskHost.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Data\Users\DefaultAccount\AppxLayouts\GrovePiVS.Debug_ARM.lambor\WinMetadata\Windows.winmd'. Module was built without symbols.
'backgroundTaskHost.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Data\Users\DefaultAccount\AppxLayouts\GrovePiVS.Debug_ARM.lambor\GrovePi.dll'. Symbols loaded.
'backgroundTaskHost.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Data\Users\DefaultAccount\AppxLayouts\GrovePiVS.Debug_ARM.lambor\System.Runtime.InteropServices.WindowsRuntime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'backgroundTaskHost.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Data\Users\DefaultAccount\AppxLayouts\GrovePiVS.Debug_ARM.lambor\System.Threading.Tasks.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'backgroundTaskHost.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Data\Users\DefaultAccount\AppxLayouts\GrovePiVS.Debug_ARM.lambor\System.Runtime.WindowsRuntime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
The thread 0x208 has exited with code 0 (0x0).
The thread 0xab0 has exited with code 0 (0x0).
Exception thrown: 'System.Runtime.InteropServices.COMException' in GrovePi.dll
WinRT information: Unexpected number of bytes was transferred. Expected: '. Actual: '.
Exception thrown: 'System.Runtime.InteropServices.COMException' in Driver.winmd
WinRT information: Unexpected number of bytes was transferred. Expected: '. Actual: '.
An exception of type 'System.Runtime.InteropServices.COMException' occurred in Driver.winmd but was not handled in user code
WinRT information: Unexpected number of bytes was transferred. Expected: '. Actual: '.
Additional information: The request could not be performed because of an I/O device error.

Unexpected number of bytes was transferred. Expected: '. Actual: '.

The thread 0xccc has exited with code 0 (0x0).
The program '[560] backgroundTaskHost.exe' has exited with code 1 (0x1).
</pre>

Anyone got this exception?

dougseven, 2016-03-24:
Just saw the same thing trying to run a Blinky project on a newly imaged RPi2

```
'backgroundTaskHost.exe' (CoreCLR: DefaultDomain): Loaded 'c:\data\Programs\WindowsApps\Microsoft.NET.CoreRuntime.1.0_1.0.23430.0_arm__8wekyb3d8bbwe\mscorlib.ni.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'backgroundTaskHost.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Data\Users\DefaultAccount\AppxLayouts\Blinky-uwpVS.Debug_ARM.dseven\Blinky.winmd'. Symbols loaded.
'backgroundTaskHost.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Data\Users\DefaultAccount\AppxLayouts\Blinky-uwpVS.Debug_ARM.dseven\System.Runtime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'backgroundTaskHost.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Data\Users\DefaultAccount\AppxLayouts\Blinky-uwpVS.Debug_ARM.dseven\WinMetadata\Windows.winmd'. Module was built without symbols.
'backgroundTaskHost.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Data\Users\DefaultAccount\AppxLayouts\Blinky-uwpVS.Debug_ARM.dseven\System.Runtime.InteropServices.WindowsRuntime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'backgroundTaskHost.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Data\Users\DefaultAccount\AppxLayouts\Blinky-uwpVS.Debug_ARM.dseven\GrovePi.dll'. Symbols loaded.
'backgroundTaskHost.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Data\Users\DefaultAccount\AppxLayouts\Blinky-uwpVS.Debug_ARM.dseven\System.Threading.Tasks.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
'backgroundTaskHost.exe' (CoreCLR: CoreCLR_UWP_Domain): Loaded 'C:\Data\Users\DefaultAccount\AppxLayouts\Blinky-uwpVS.Debug_ARM.dseven\System.Runtime.WindowsRuntime.dll'. Skipped loading symbols. Module is optimized and the debugger option 'Just My Code' is enabled.
The thread 0xd94 has exited with code 0 (0x0).
The thread 0xfa0 has exited with code 0 (0x0).
The thread 0xd24 has exited with code 0 (0x0).
Exception thrown: 'System.Runtime.InteropServices.COMException' in GrovePi.dll
WinRT information: Unexpected number of bytes was transferred. Expected: '. Actual: '.
An exception of type 'System.Runtime.InteropServices.COMException' occurred in GrovePi.dll but was not handled in user code
WinRT information: Unexpected number of bytes was transferred. Expected: '. Actual: '.
Additional information: The request could not be performed because of an I/O device error.

Unexpected number of bytes was transferred. Expected: '. Actual: '.
```

DexterInd, 2016-04-25:
@dougseven , any idea why this happened?

DexterInd, 2016-11-05:
Just tested this on a Pi 3 and OS 10.0.14393.0 seems to work fine, no error thrown.

closed 2016-11-05 by DexterInd