DexterInd/GrovePi #237: Multiple Relay

Zangror, 2016-04-27:
We've bought three [relays](http://www.seeedstudio.com/wiki/Grove_-_SPDT_Relay%2830A%29), and we use the following Python Class to interact with the relay (see attached file). Nothing complicated. It's just Python 3 and more object encapsulation.

The problem we have is that when we set off a relay, the two other relays also set off ! Which is not what we're expected.

For example:

    r1 = Relay(5)
    r2 = Relay(6)
    r3 = Relay(7)
    
    r1.on()
    r2.on()
    r3.off() # r1 and r2 are now off ...

We observed the same problem without the object encapsulation. And thus we suspect that the problem is coming from GrovePi.

Is it a know bug? Or is GrovePi intended to only work with one relay?

[relay.py.txt](https://github.com/DexterInd/GrovePi/files/238412/relay.py.txt)

karan259, 2016-04-27:
Hey @Zangror : we'll try the example that you have sent and see how it works at our end.

Zangror, 2016-04-27:
@karan259 : thanks! :smiley: 

It could also be an hardware relay problem...

karan259, 2016-04-27:
do you have some LED's . The if the LED's work on the same ports, then it might be a hardware problem. (the same code should work with the LED's too)

Zangror, 2016-04-27:
By led you mean: http://www.seeedstudio.com/wiki/Grove_-_LED this kind of led?

karan259, 2016-04-27:
yeah, those ones. (Logically to the code they are the same)

Zangror, 2016-04-27:
No we don't have any leds...
We only have three [relays](http://www.seeedstudio.com/wiki/Grove_-_SPDT_Relay%2830A%29), one [barometer sensor](http://www.seeedstudio.com/wiki/Grove_-_Barometer_Sensor) and one [temperature and humidity sensor](http://www.seeedstudio.com/wiki/Grove_-_Temperature%26Humidity_Sensor_%28HDC1000%29) on the same GrovePi+ .

karan259, 2016-04-27:
Ok. I'll try it out at my end and get back to you

Zangror, 2016-04-27:
Thanks in advance! :smile:

karan259, 2016-04-29:
Hey @Zangror : I just tried out your test code and it does work properly. I only had 2 relays so used that with an LED and it worked properly. Can you try out this 
[relay_test_py.txt](https://github.com/DexterInd/GrovePi/files/241778/relay_test_py.txt). Can you also run https://github.com/DexterInd/GrovePi/blob/master/Software/Python/grove_firmware_version_check.py and check if the firmware on the GrovePi is v1.2.2

Zangror, 2016-04-29:
Our firmware version is v1.2.7.

Thanks for the feedback !

In our case, we are looping several times on the relay switching. Is it possible to you to run the following code and see if there is any illogical behavior?

    import time
    
    for _ in range(0, 10):
        r8 = Relay(8)
        r7 = Relay(7)
        r5 = Relay(5)
        time.sleep(1)
        r8.on()
        r7.on()
        time.sleep(2)
        r8.off()
        r5.on()
        r7.off()
        time.sleep(2)
        r5.off()

karan259, 2016-05-09:
Hey @Zangror : this script also works for me. I am not sure where the 1.2.7 firmware came from. The latest stable version that we have is 1.2.2 and the beta is also at 1.2.6. Can you try updating the firmware from here: http://www.dexterindustries.com/GrovePi/get-started-with-the-grovepi/updating-firmware/ and check the firmware version and try your example again

karan259, 2016-05-23:
@Zangror Did you get a chance to try this out