DexterInd/GrovePi #241: avrdude: Can't find programmer id "gpio"

fbrier, 2016-05-01:
Based on a [post on the forums](http://www.dexterindustries.com/topic/grovepi-not-working-problem-updating-firmware/), this may be occurring because I installed the lastest version of Raspbian (Jessie).  The blinking LED test program did not work, so I tried to upgrade the firmware and received the 'avrdude: Can't find programmer id "gpio"' error.  Then I tried running the Troubleshooting program and received the same error.  The above post mentioned [work on fixing this problem was in beta for Raspian for Robots](http://www.dexterindustries.com/topic/jessie-raspbian-for-robots/).  When might those changes find their way into this GrovePi project?

karan259, 2016-05-02:
@fbrier : Looks like you are using a fresh Jessie image. Did you use our install scripts to set everything up on the Pi. If so can you send a link to the install scripts that you had used. Also, can you upload the log form the troubleshooting script

fbrier, 2016-05-02:
Yes, I am using the latest Raspian image (2016-03-18-raspbian-jessie.zip).  The installation script was just cloned from this git repo.  Attached is the [GrovePiTroubleshootingLog.txt](https://github.com/DexterInd/GrovePi/files/244651/GrovePiTroubleshootingLog.txt).

karan259, 2016-05-09:
Hey @fbrier : can you try running this script https://github.com/DexterInd/ArduBerry/blob/master/script/install.sh and check if it works for you.

nzmarkc, 2016-05-15:
I'm having a similar problem to @fbrier with my GrovePi. It is the grove_button.py that isn't working for me, the blinking LED program did work though. I'm running on a Pi3, my OS details are:

- Linux raspberrypi-mtc 4.4.9-v7+ #884 SMP Fri May 6 17:28:59 BST 2016 armv7l GNU/Linux
- Linux version 4.4.9-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #884 SMP Fri May 6 17:28:59 BST 2016
- Distributor ID: Raspbian
- Description:    Raspbian GNU/Linux 8.0 (jessie)
- Release:        8.0
- Codename:       jessie

I tried checking my firmware version via the firmware_version_check.py in the Troublshooting directory and it gives me: 

- GrovePi has firmware version: 255.255.255

Which in http://www.dexterindustries.com/topic/grovepi-firmware-update-v1-2-2/ karan mentions: "If it returns something like 255.255.255, then you are probably running a very old firmware. Try running the script a couple of times." which I did and I get the same message every time. I have attached my troubleshooting log fyi.

[grovepi-trouble-log-1.txt](https://github.com/DexterInd/GrovePi/files/264939/grovepi-trouble-log-1.txt)

It would be great to sort this out as soon as possible. Thanks for the help!

CleoQc, 2016-05-15:
I was able to get it to work on a fresh Jessie.
You need to run the following two scripts:
https://raw.githubusercontent.com/CleoQc/Raspbian_For_Robots/master/upd_script/update_GrovePi.sh
https://raw.githubusercontent.com/DexterInd/Raspbian_For_Robots/master/upd_script/update_GrovePi_Firmware.sh

The first one is in my personal repo, because there was a typo in the original script. The PR is already sent and will likely be pulled in within the next day (it's just a typo)

nzmarkc, 2016-05-16:
Thank you Cleo. The update_GrovePi.sh runs fine. But the the update_GrovePi_Firmware.sh gives me the following error:
**********
avrdude: AVR device not responding
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.
**********
Not sure what to do at this stage. I think I might have a faulty GrovePi. :-(

fbrier, 2016-05-16:
Hi karan259!  Your recommendation worked.  I pulled down the ArduBerry repo and ran its install.sh.  The install.sh log only shows one odd message:
```
Error: Could not open file `/dev/i2c-0' or `/dev/i2c/0': No such file or directory
```
However, the output for checking the I2C bus 0 seems to be correct.  The GrovePi then reported the firmware version as 255.255.255.  The grove_led_blink.py program now causes the LED to blink properly.  So then I went to the GrovePi repo and ran the firmware_update.sh and it seemed to run perfect.  The firmware_version_test.sh reports the firmware version as 1.2.2.  So other than that odd message about not being able to open /dev/i2c-0, everything seems to be functioning and now I can play with all my different sensors.  Thank you very much.

Sorry for not getting back sooner.  I had another Raspberry Pi with an Adafruit breakout hooked up to my KVM and was trying to get an IR remote using LIRC working for a user group meeting.  Didn't make it, but I tried.  Again, thank you.

closed 2016-05-16 by fbrier

matbeerkens, 2016-05-19:
Hi all, I also encountered this error. As it turned out avr was not correctly installed, so i did the following: upon login i changed to root user with `sudo su`. Next i went ito direct Scripts like `cd Desktop/GrovePi/Script` then i ran install.sh like `bash install.sh`. That took some time to get processed after which avr gave me the correct response. This can be tested like `avrdude -P gpio -c gpio -p atmega328p` The response should contain 

> avrdude: safemode: Fuses OK

.

nzmarkc, 2016-05-19:
Thank you Mat - but no go I'm afraid. I followed your instructions and the 'bash install.sh' ran fine. What I get from the test instruction is:

**********
avrdude: AVR device not responding
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.
**********
I have thread going on Dexter Industries (http://www.dexterindustries.com/topic/grovepi-firmware-update-v1-2-2/page/2/#post-91442) but haven't heard back yet. Wondering if I have a faulty GrovePi.

Thanks again for the help.

karan259, 2016-05-20:
Hey @nzmarkc: that error means that avrdude is working fine but some problem with the board. Can you reach out on the forums. Would be easier to help you out there

nimish, 2016-08-29:
Does this programmer work for the RPi3 then? It's all that seems to matter in the install.sh, which installs an ancient version of avrdude

`programmer
  id    = "gpio";
  desc  = "Use sysfs interface to bitbang GPIO lines";
  type  = gpio;
  reset = 8;
  sck   = 11;
  mosi  = 10;
  miso  = 9;  
;
`

karan259, 2016-08-30:
Hey @nimish : this still does work with the Pi3. We have tested this and it works pretty well with all our products. Are you having a specific problem with this

nimish, 2016-08-31:
No, I just want to confirm that all I need is the programmer description: i'd like to move to a later version of avrdude

dothebart, 2018-02-20:
Hm, a year later, and this change still wasn't pushed upstream. Only https://github.com/DexterInd/AVRDUDE will contain the proper settings.