DexterInd/GrovePi #251: No data if using 2 digital sensors

cde93, 2016-05-25:
If I use 2 digital sensors, one of the sensors gets no data.

DexterInd, 2016-05-26:
Can you help us understand the problem a little more?  What sensors, what code?

cde93, 2016-05-26:
My description was too briefly, I apologize.
If I use the Temperature and Humadity Sensor and short after an other digital sensor this sensor gives only zero back. But after updating to the latest version that changed. The sensor gives alway's zero as output for both Temperature and Humadity. 
So I have checked the source code for reading that sensor, en stepped through it with the debugger. There is an method called Measure(), witch actually reads the data from the sensor, but he's never called? If I call it before accessing the sensor data it also gives 0. If I add the method to the specified getters it works as expected. 
(Sorry for my bad english, it's not my native language)

karan259, 2016-05-26:
our forums might be a better place to resolve this. can you post a link to the code that you are using and a picture of your setup in the forums here: http://www.dexterindustries.com/forum/grovepi/. Are you using Dexter Industries Raspbian for Robots image on the SD card or something else

cde93, 2016-05-26:
No, i'm using Windows iot Core with the C# lib. I think it's an bug in the lib, that is why I posted it here.
`var a = DeviceFactory.Build.DHTTemperatureAndHumiditySensor(Pin.DigitalPin7, GrovePi.Sensors.DHTModel.Dht11).TemperatureInCelsius;`
Gives a 0 result, even if I first call:
`DeviceFactory.Build.DHTTemperatureAndHumiditySensor(_pinHumanityAndTemperature, GrovePi.Sensors.DHTModel.Dht11).Measure();`