DexterInd/GrovePi #261: Motor Shield v1.3 - python issue

crazyquark, 2016-06-28:
Hello,

I am having an issue with the motor shield on grovepi running in Python.
I built a small robot and every once in a while the motor shield stops responding. Resetting doesn't help and I have to reboot the raspberry pi to get it working again.

I power the motors using 6 battery pack while the rpi is powered independently using an usb battery.
(jumper is removed)
The error I get is IOError(110).

I wish I knew how to debug this... I originally thought that it could be a wire connection problem but it's not.
After resetting the driver I can see it with i2cdetect -y 1 but after the IOError(110) it vanishes again.
I am looking for debugging tips mostly.

Thank you

DexterInd, 2016-06-29:
Which motor shield are you running?

crazyquark, 2016-06-29:
This one: [http://www.seeedstudio.com/wiki/Grove_-_I2C_Motor_Driver_V1.3](1.3)
The PCB label is however inscribed as v1.3b, not sure if there is a difference

DexterInd, 2016-06-29:
Hmm.  The shield may be locking up the I2C line, either pulling it high or pulling it low.  One small option may be to remove the pullups on the the motor shield.  You might need a logic analyzer to see if the lines are getting pulled high or low.  

Do you see the same behavior with just the GrovePi attached?

crazyquark, 2016-06-29:
This only happens with the motor controller. After reseting I can see it with i2cdetect -y 1 but if I try to connect to it times out and messes up the i2c bus again.

DexterInd, 2016-07-01:
You might want to try to hook up a logic analyzer or an oscope to it; it might be drawing too much power and causing the pi to crash.  When you run the motors really high, it might be causing a voltage drop?

crazyquark, 2016-07-13:
But the Pi is not crashing, only the i2c interface loses connection to the motor shield.

crazyquark, 2016-10-13:
I know this is an old issue but I haven't gotten around to further debug this.
After the motor shield crashes I can see in the kernel log:
 91.644852] i2c i2c-1: transfer timed out
[   92.644861] i2c i2c-1: transfer timed out
[   93.644869] i2c i2c-1: transfer timed out
[   94.644877] i2c i2c-1: transfer timed out
[   95.644885] i2c i2c-1: transfer timed out
[   96.644894] i2c i2c-1: transfer timed out
[   97.644903] i2c i2c-1: transfer timed out
[   98.644910] i2c i2c-1: transfer timed out
[   99.644918] i2c i2c-1: transfer timed out
[  100.644927] i2c i2c-1: transfer timed out
[  101.644949] i2c i2c-1: transfer timed out
[  102.644943] i2c i2c-1: transfer timed out
[  103.644951] i2c i2c-1: transfer timed out
[  104.644961] i2c i2c-1: transfer timed out
[  105.644970] i2c i2c-1: transfer timed out
[  157.845408] i2c i2c-1: transfer timed out

crazyquark, 2016-10-13:
I think I might have found a workaround. By limiting the speed of the motors to < 100 (I am using 95% now) it seems I can avoid crashing the motor shield and jamming the i2c line.

closed 2016-10-18 by crazyquark