DexterInd/GrovePi #296: Relays are not switched

volkf, 2016-12-21:
I have this code:

```while True:
    try:
        grovepi.digitalWrite(relay0,0)
        grovepi.digitalWrite(relay1,1)
        time.sleep(5)
        grovepi.digitalWrite(relay1,0)
        grovepi.digitalWrite(relay0,1)
    except(IOError):
        print("Error")
```
This code does not work for me. One relay will be switched and the other is not switched.
If i run just run:

```
grovepi.digitalWrite(relay0,0)
grovepi.digitalWrite(relay1,1)

```
Then it works. What is the problem?

CleoQc, 2016-12-21:
Could you add a sleep(5) at the end of the loop too? 
What's happening is that it's looping too fast. 
It goes A->sleep->B->A->sleep->B->A->sleep

volkf, 2016-12-21:
Thanks, yes I see. The five seconds delay at the end solve the problem in this example.
But it raises a new question: Whats the minimum time between the switch command?

This is the original code:

```
 if oldTemperature >= setpoint+0.25:
      grovepi.digitalWrite(relay0, 0)
      time.sleep(0.5)
      grovepi.digitalWrite(relay1, 1)
      logging.info("Close")
  elif oldTemperature < setpoint-0.25:
      grovepi.digitalWrite(relay1, 0)
      time.sleep(0.5)
      grovepi.digitalWrite(relay0, 1)
      logging.info("Open")
  else:
      grovepi.digitalWrite(relay0, 0)
      time.sleep(0.5)
      grovepi.digitalWrite(relay1, 0)
      logging.info("Neutral")
  time.sleep(1)
```
The problem here was that relay A is switched (red led on). Then the other relay B is switched (red led on). But at this point relay A is not switched anymore (red led still on).

CleoQc, 2016-12-21:
Is that the old code? And which relay is A or B, (relay0 is A?)

volkf, 2016-12-21:
That code is what I want to achieve with 2 relays. Yes relay0 is A in my description.

```
grovepi.digitalWrite(relay1, 0)
time.sleep(0.5)
grovepi.digitalWrite(relay0, 1)
```
The first line is completely ignored...relay0 is open but relay1 is still closed (red led on).
I hope this makes the problem clearer to you :smiley:

CleoQc, 2016-12-21:
I have it working here with two relays and a delay of .5 second without any issue. So the issue is elsewhere.
Can you send me your whole code at nicole@dexterindustries.com ?

closed 2017-12-07 by volkflo