DexterInd/GrovePi #313: Plan to add 5 Grove Driver in C#

awong1900, 2017-02-25:
I will add 5 Grove Driver, this:
PIR Sensor, 
Gas Sensor,
3-Axis Digital Accelerometer(±16g), 
OLED 96x96 v2.1,
Mini I2C Motor Driver v1.0

Then can you update the GrovePi C# library on Nuget?

DexterInd, 2017-02-26:
Yes, we will try to update the Nuget!

awong1900, 2017-05-12:
@DexterInd Do you have updated Nuget library?  I found the v1.0.9 not include Mini I2C Motor Driver.

DexterInd, 2017-05-12:
Hey @awong1900 can you try this?  Just want to make sure it works for you before we release it.  [Here's the link to v1.0.10](https://www.nuget.org/packages/GrovePi/1.0.10).

DexterInd, 2017-05-15:
Hey Ten just wanted to see if you had tried the updated nuget?

awong1900, 2017-05-15:
Yes, I am trying test it.

awong1900, 2017-05-15:
Hi @DexterInd It work. Please release it.

DexterInd, 2017-05-15:
Great!  It is now released!

closed 2017-05-15 by DexterInd