DexterInd/GrovePi #321: Scratch: Sensor value types aren't selectable until reports come in

fhunleth, 2017-02-27:
This turned out to be a source of confusion for kids and mentors at our CoderDojo club. They would have sensors plugged in and due to a typo in the corresponding broadcast request message or not paying attention, they'd struggle with adding a sensor value block with the correct type. Many non-GrovePi+ sensors show up in the pulldown menu and this made people think that some sensors worked and some didn't.

I'm wondering if reporting all sensor values at startup for all GrovePi+ sensors would be helpful. No real interaction with hardware would be done. For example, 0 could be reported up for every possible sensor type. That way users wouldn't have to double click a broadcast block to prime the sensor list every time they start Scratch. The downside is that the list might be long because you'd have things like `button2` to `button8`.

I'm inclined to make this change but wanted to get the community's feedback if there was an issue with this approach.

CleoQc, 2017-04-19:
Sorry it took a while to come back. I totally agree with you that it's more than annoying. However the number of sensors would make a very long list, especially since many can go on different ports and each combination would have to be listed. 
I had it done and working at one point and it wasn't really working, the list was just too long when you wanted to select a sensor.

fhunleth, 2017-04-19:
Yep - I was worried about the really long list. I wish there were another way, since the issue popped up again last meeting. I'll post if I get other ideas. Thanks for the response.

closed 2017-04-19 by fhunleth

CleoQc, 2017-04-19:
The way I've presented it to kids is to have them build a "setup" sequence.  They drag broadcast blocks to it corresponding to all their sensors, and then click on that sequence in order to "tell" their robot what the sensors are. (all it really does is activate the sensor list accordingly)