DexterInd/GrovePi #327: Firmware Update - GrovePi must be cloned to the Desktop to update firmware.

ecarlisle, 2017-03-27:
Hi!
First of all, thank you for the GrovePi! I love it! Initially I could not update the firmware and discovered that is was because of a hardcoded path in [firmware_date.sh](https://github.com/DexterInd/GrovePi/blob/master/Firmware/firmware_update.sh#L27).  I cloned the repo into `/home/pi/products` but this doesn't work with the source path set for `grovepi_firmware_update.sh`.

```
source /home/pi/Desktop/GrovePi/Firmware/grovepi_firmware_update.sh
```

I changed the path to...

```
source ./grovepi_firmware_update.sh
```

... and the firmware update completed successfully.

```
...
avrdude: verifying ...
avrdude: 13816 bytes of flash verified
avrdude: safemode: Fuses OK
avrdude done.  Thank you.
```

Can this source path be changed so that there's no dependency on the clone repo path?

**Caveat:** I'm by no means a shell script expert, so there might be a good reason the source path is set as such.  If it can be changed, I'm happy to submit a PR.

Again, thank you for the GrovePi! I just bought a couple of GrovePi Zeros for a robot project I'm working on.  Can't wait to start working with them!

Best!
Eric

Shoban94, 2017-03-31:
Hi @ecarlisle ,  We have recently made a commit here https://github.com/DexterInd/GrovePi/pull/328 that removes almost all of the dependencies on the Desktop folder and you can now clone GrovePi to any folder and have it working. Please let us know if that helps.
-Shoban

ecarlisle, 2017-04-01:
Will do. Thank you very much for your help!

closed 2017-04-01 by ecarlisle