DexterInd/GrovePi #335: Temprature&Humidity Sensor wrong values

minhajakhterkhan199, 2017-04-19:
Dear,
We are using Temprature&Humidity v1.2, it is giving us wrong values like Temperature = 563.2  Humidity = 691.2. 
Please find our source code below from golang. Please advise further to solve the issue
	//Temprature & Humity Sensor
	var g grovepi.GrovePi
	g = *grovepi.InitGrovePi(0x04)
	for {
		time.Sleep(2 * time.Second)
		t, h, err := g.ReadDHT(grovepi.D4)
		if err != nil {
			fmt.Println(err)
		}
		fmt.Println("Temperature =",t," Humidity =", h)
	}
	g.CloseDevice()

Regards,

minhajakhterkhan199, 2017-04-27:
Please respond on above query.

RobertLucian, 2017-04-28:
Issues on GitHub are meant to notify the team of a `bug` /  `proposed enhancement` / `task`. 
The idea behind of it, is to create an issue when you know for sure what the problem / task is.  And for now, this isn't the case.

But on the other hand, you seem to have a problem with your code. I'll ask you kindly to post this problem on the forum and we will try to help you out.
The more you help us, like with code indentation or screenshots, the faster we can solve the issue.

And also, please confirm you're using nodeJS on the forum.

Thank you!

minhajakhterkhan199, 2017-04-30:
Thanks for your response. 
This sample code has been taken from same repository, link is down below 
https://github.com/DexterInd/GrovePi/blob/master/Software/Go/grovepi_dht.go
We are using golang. 
In this regards, waiting for your response.

Regards,

LuCavallin, 2017-06-15:
Same issue here, I get:

T: 716.799988 - H: 1638.400024

LuCavallin, 2017-06-15:
Problem solved!
This line in ReadDHT

`b := []byte{CommandDHTRead, pin, 1, 0}`

change it to

`b := []byte{CommandDHTRead, pin, 0, 0}`

as per http://forum.dexterindustries.com/t/how-to-convert-interpret-dht-sensor-to-human-readable-values-solved/2144/4

closed 2017-12-07 by johnisanerd