DexterInd/GrovePi #336: PyPi package is very, VERY out of date.

fake-name, 2017-05-09:
I'm trying to use a GrovePi for a fairly complex existing project, to provide some convenience IO. As such, I'm not using any of the provided OS images.

At the moment, there seems to be no easy way to manually install the GrovePi library:

 - PyPi is out. 
 - Repository format is not correct for `pip install git+https://github.com/DexterInd/GrovePi.git`
 - Manual git-cloning, and running `setup.py` fails because there are dependencies missing from `setup.py`.

Basically, how are you /supposed/ to install this thing?

CleoQc, 2017-05-09:
Clone repo and run this script:
https://github.com/DexterInd/GrovePi/blob/master/Script/install.sh

fake-name, 2017-05-09:
Ok, so I need `smbus`, which is actually `smbus-ffi`. That depends on the `libffi-dev` apt-package, and having `build-essentials` (which I already had).

fake-name, 2017-05-09:
>> `curl --silent https://raw.githubusercontent.com/DexterInd/script_tools/master/install_script_tools.sh | bash`

**Really**?

closed 2017-05-09 by fake-name

fake-name, 2017-05-09:
Other stuff:

```
sudo apt-get purge python-rpi.gpio -y
sudo apt-get purge python3-rpi.gpio -y
sudo apt-get install python-rpi.gpio -y
sudo apt-get install python3-rpi.gpio -y
sudo pip install -U RPi.GPIO
```

Uh, what? You're installing the apt-package, and then installing the PyPi package on top of it? That's a terrible idea, and probably a great way to wind up with broken packages, particularly if/when apt decides to overwrite parts of the files managed by pip.