DexterInd/GrovePi #345: grove CO2 Sensor not reading any value

sahanasadagopan, 2017-06-07:
Hi,
I have connected grove co2 sensor with grove pi+ board, I am stuck with getting value for the co2 sensor. I tried to read and read through the uart pin between two pi;s and it works fine. but when I am trying to use the grove C02 and read the conc value ,I am not able to read any value.
Could some one help me with the issue?

RobertLucian, 2017-06-26:
I can read the `conc` value just fine - so no issues on my side.
We should be running the same code on your `Raspberry Pi` in order to pinpoint the issue.

So could you run the following code (run it from within `Python/grove_co2_sensor` directory) and tell us what it's the output of it?
```
from grove_co2_lib import CO2

co2_sensor = CO2()

for step in range(10):
    conc = co2_sensor.read()[0]
    print("conc = " + str(conc))
```

This is what I get when I run the program.

[image]

Can you confirm all of the above?

Thank you!

RobertLucian, 2018-07-05:
Might need to further investigate this.

RobertLucian, 2018-07-27:
I have found out that sometimes the Raspberry Pi's serial port is not working, though it's enabled. I have placed the logic analyzer on the serial pins and within Python I'm not able to write or read anything from serial port when this happens. Basically the program just hangs on the write function. 

When the serial port is working, so does the program. Therefore, there's nothing wrong with the Python script for the CO2 sensor, but with the serial port of the Pi.

closed 2019-04-27 by RobertLucian