DexterInd/GrovePi #360: GrovePi disconnect from I2C on kernel 4.9.35

KJee85, 2017-07-19:
Recently updated some devices using GrovePi+ this moved them to the new 4.9.35 kernel. I also then pulled down the latest of this repo and flashed the latest firmware 1.2.7 to the GrovePi.

Everything appears to work normally. I can see the Grove pi via i2cdetect -y 1 and also get data from my sensors attached. After some time it no longer appears via i2cdetect and I get errors on reads.

I've found I can bring this back by cycling gpio pin 10 (CE0) then the GrovePi shows back up but some time later it will happen again.

Here is some hopefully relevant info.

Running dmesg I see:
`i2c-bcm2835 3f804000.i2c: i2c transfer timed out`

Running lsmod:
```
Module                  Size  Used by
binfmt_misc             7988  1
bnep                   12051  2
cfg80211              543027  0
evdev                  12423  1
snd_usb_audio         147900  0
snd_hwdep               6688  1 snd_usb_audio
snd_usbmidi_lib        22487  1 snd_usb_audio
snd_rawmidi            23790  1 snd_usbmidi_lib
snd_seq_device          5504  1 snd_rawmidi
snd_bcm2835            24427  0
snd_pcm                98501  2 snd_usb_audio,snd_bcm2835
snd_timer              23968  1 snd_pcm
snd                    70032  8 snd_hwdep,snd_usb_audio,snd_timer,snd_rawmidi,sn               d_usbmidi_lib,snd_seq_device,snd_bcm2835,snd_pcm
bcm2835_gpiomem         3940  0
i2c_bcm2835             7167  0
btusb                  30139  0
btrtl                   5500  1 btusb
btintel                10199  1 btusb
btbcm                   7916  1 btusb
bluetooth             365511  26 btrtl,btintel,bnep,btbcm,btusb
rfkill                 20851  3 bluetooth,cfg80211
uio_pdrv_genirq         3923  0
fixed                   3285  0
uio                    10204  1 uio_pdrv_genirq
i2c_dev                 6913  0
ipv6                  408971  32
```
I don't see any other info in logs around i2c but maybe I'm just not looking in the right place. One other thing I noticed was initially after upgrade both i2c_bcm2835 and the older i2c_bcm2708 as we had an entry for i2c-bcm2708 in /etc/modules which is how it was working in 4.1.19. Even after removing this and rebooting the issue remains. Let me know any other info / steps I can take to further identify this issue.

DexterInd, 2017-07-20:
We have a known issue with the latest kernel at the moment, I'm not sure if it affects what you're working with.  You can see the detailed conversation about it here (https://github.com/raspberrypi/firmware/issues/828).  This has to do with handling repeated start/stops in I2C.  The workaround we are currently using on all our software is to roll the kernel back.  

You can roll your kernel back with this command: `  sudo rpi-update 52241088c1da59a359110d39c1875cda56496764  # kernel: Bump to 4.4.50 - v7+`

Just to dive a little deeper though: this should only affect the `bus.read_i2c_block_data` command, which with the GrovePi affects a few sensors, including the analog read.  Does this sound like what you're seeing?

KJee85, 2017-07-20:
That does seem about right. We are doing analog, digital and DHT reads. 

Rather than roll all the units back do you think my workaround for resetting the GrovePi via CE0 is viable as well? This does seem to have it show back up on I2C.

DAnGioi, 2020-03-29:
Hello, I have the same problem.
My kernel : Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
Have you solved your problem?