DexterInd/GrovePi #364: Can't get GPS module working

AlexRex, 2017-07-28:
I have a GPS module installed in a GrovePI, using the RPISER port. 

Running the python script GroveGPS.py I get the following output: 

```console
smartua@smartuapi:~/GrovePi/Software/Python/grove_gps $ sudo python GroveGPS.py

Time: 145637.092 Fix status: 0 Sats in view: 0 Altitude  Lat:   Long:
Raw String appears to be empty.
```

I tried enabling and disabling serial ports from raspi-config, but I cannot get it working.

DexterInd, 2017-07-28:
Does the time increment?  As you read it repeatedly, do you see the time numbers change?

AlexRex, 2017-07-28:
@DexterInd Yes, I see how it increments.

DexterInd, 2017-07-28:
That means it should be working, you just need to go outside and catch a GPS satellite or two.   You have 0 Sats in view right now, you'll need 3-4 to get a good GPS position.

AlexRex, 2017-07-28:
I done what you say and now I am getting a lot of sats available, but the output it's quite weird: 

```console
smartua@smartuapi:~/GrovePi/Software/Python/grove_gps $ python GroveGPS.py

Time: 162201.000 Fix status: 1 Sats in view: 10 Altitude 0.9
                                                            M Lat: 3812.5028 N Long: 00001.4634 W

Time: 162251.00 Fix status: 10 Sats in view: 0.91 Altitude M Lat: 3810.4328  Long: 00041.464
                                                                                            G 1
Raw String appears to be empty.

Time: 16225.000 812.308 Fix status: 0 1 Sats in view: 21 9 M Altitude 08
                                                                        b�bN�b��b��`��b�`��	 �I�`90 Lat: N 0041.4634 Long: W 1 1
Raw String appears to be empty.

Unable to read

Time: 126.000 Fix status: 32 Sats in view: 00 Altitude 41 Lat: 3812.58 @ Long: 0004.4630������
r�
���H�5)�M���b�b�`�2 01
Raw String appears to be empty.

Unable to read

Time: 162630.00 Fix status: 10 69028 Sats in view: 39 20
                                                         Altitude 08 Lat: 0.5328 N
                                                                                  00000 4630
                                                                                            P
                                                                                             1 Long: �)�b�
b
b��r�����I	5(�AM�����`�`��b���b����b��b����b`�r��b�rʊ`�r��jB PPF

                                                                     1 11
Raw String appears to be empty.

Time: 0Sr�`���p���br1�@�r���"`��ű���rb�p�bj���9�`����1)�AM��4Ċ8 Fix status: 1 Sats in view: 20 Altitude 0.24 0.91
                                                                                                                  0.R0
                                                                                                                     $GPG@F Lat: 013 22 Long: 08 04 3
Raw String appears to be empty.

Unable to read

Unable to read

Unable to read

Unable to read

Unable to read

Time: 1600100 Fix status: 11 Sats in view: 71
                                             331
                                                00
                                                  22 Altitude 1 Lat: 380.43 N
                                                                             00041���)b�b��)�b��r�bjb�r�b.��I�j@ :A1ǚb���b��`�b��b��b��b��bbb	p�b�ʲ*�R

$@@GCF 3 Long: 1 12
Raw String appears to be empty.

Unable to read

Time: 163259.00 Fix status: 0 Sats in view: 9 0.9 Altitude M Lat: 3812.5328 @ Long: 00041.4034 W

Unable to read

Time: 163445000 80 532 @ Fix status: 331 Sats in view: 4 Altitude 67 230
                                                                        8�b���`��b��b*�� �"�@�
jR"AMYǚb�b
�b�`��b���I�b���(08 Lat: 0��r����`�b

b��)�`��r�`j���ٱ��bB:R"u5��ib
���b�
`��b�"b���b�bbb�(r��`�r�2b�r��R�jP P@@ 3 0 Long: 2 11 73
Raw String appears to be empty.
```

I also tried with the example `dextergps.py` and I am getting this output:

```console
smartua@smartuapi:~/GrovePi/Software/Python/grove_gps $ python dextergps.py
['$GPGGA', '100950\x000\xc0\x81\xc1\xe1\xc5\xc9\xb8\xd5\x01\xc9\x01\xb19\xb0\xc1\xc1\x07D\x8ar \x80\x9a"b\xbab\x82b\x8a\x00\x0ep\xc8\xb2`\x90\x82\x02\x02\x00']
Failed: wrong number of parameters
['$GPGGA', '100950\x000\xc0\x81\xc1\xe1\xc5\xc9\xb8\xd5\x01\xc9\x01\xb19\xb0\xc1\xc1\x07D\x8ar \x80\x9a"b\xbab\x82b\x8a\x00\x0ep\xc8\xb2`\x90\x82\x02\x02\x00']
```
Which is also really weird. Am I doing something wrong here?

RobertLucian, 2017-07-29:
@AlexRex I will test `GroveGPS.py` script and see if I can repeat your issue.

AlexRex, 2017-07-29:
@RobertLucian Thanks Robert. It's quite frustrating because I bought three of them and none is working correctly 😢

RobertLucian, 2017-07-31:
@AlexRex I've been testing the GPS module and I haven't encountered your issue (yet). The sats are in the view and it spits the right strings/values.

It seems like there's an encoding issue somewhere.
I'm also wondering if the `readlineCR` works as intended.

I can think of a couple of things for now:

1. Replace [line 63](https://github.com/DexterInd/GrovePi/blob/master/Software/Python/grove_gps/GroveGPS.py#L63) in `GroveGPS.py` file with the following line of code:
`print(repr("Time: {} Fix status: {} Sats in view: {} Altitude: {} Lat: {}{} Long: {}{}".format(t,fix,sats,alt,lat,lat_ns,long,long_ew)))`
By using `repr` function, we get an ASCII string that unambigously represents what you got. It should give us a couple of more hints into solving the issue.

1. Use a Python shell or open up an empty python script where you copy-paste the following [couple of lines](https://github.com/DexterInd/GrovePi/blob/master/Software/Python/grove_gps/GroveGPS.py#L1-L19).
Then try reading with `readlineCR` function continuously the following way:
```
# copy-pasted code from GroveGPS module

while True:
    print(repr(readlineCR()))
```

3. Share the `.csv` file that has been generated while `GroveGPS.py` ran - the one where the sats were detected.

DexterInd, 2017-07-31:
@AlexRex how are you powering the Raspberry Pi and GrovePi when you have it outside?  If the voltage is low, some of the serial lines may be low-leveling the serial communications and providing the garbage in between valid readings (aside from the random characters, your data looks pretty good there).  So . . how are you powering this setup?

AlexRex, 2017-08-01:
@RobertLucian With the first option I got this output: 

```console
'Time: 152738.000 Fix status: 2 Sats in view: 11 Altitude: 15.0 Lat: 3812.5265N Long: 00041.4534W'
Raw String appears to be empty.

'Time: 152739.000 Fix status: 2 Sats in view: 11 Altitude: 15.0 Lat: 3812.5265N Long: 00041.4534W'
Raw String appears to be empty.

'Time: 152743.000 Fix status: 2 11 Sats in view: 0.78 Altitude: 50.6 Lat: 3812.5265N Long: 00041 \x04534W'
Raw String appears to be empty.

'Time: 152746.000 Fix status: 2 Sats in view: 11 Altitude: 15.0 Lat: 3812.5265N Long: 00041.4534W'
Raw String appears to be empty.

'Time: 152750.000 Fix status: 2 Sats in view: 11 Altitude: 15.0 Lat: 3812.5265N Long: 00041.4534W'
Raw String appears to be empty.
```

It looks quite better, but looking at google maps my gps coord should be this ones: https://goo.gl/maps/kTJ4D3eiCFv

And I am getting this ones: https://goo.gl/maps/oDQxJSAMVXR2
So them doesn't look right. 

With the second method I am getting this output: 

```console
....
''
''
''
''
''
' GPGGA,152524.000,3\x0012.5265,N,00041.4534,\x17,2,10,0.85,15.0,M,50.6,M,0000,0000*4C\r'
'\n$GPGSA,A\x0c3,08,32,16,28,01,10,27,11,22,14,,,1.59,0.85,1.35*01\r'
'\n$GPGSV,4,1,13,08,84,099,34,11,52,305,28,27,46,127,32,22,43,215,24*73\r'
'\n$GPGSV,4,2,13,0\x01,41,286,31,36,34,134,39,32,29,092 34,10,28,046,34*71\r'
'\n$GPGSV,4,3,13,14,25,112,40,28,18,318,26,03,17,217,14,16,08,176,34*79\r'
'\n$GPGSV,4,4,13,18,02,045,24*47\r'
'\n$GPRMC,152524.000,A,3812.5265,N,00041.4534,W,0.00,282.59,010817,,,D*08\r'
'\n'
''
''
''
''
''
....
```

There is a lot of empty strings and time to time I get info like this. 

And in the `.csv` file there's nothing, just one line with the titles: `name,latitude,longitude`.

@DexterInd I am using generic charger: 5v / 2000mAh. Should it be more?

Thank you both! I really appreciate it!

AlexRex, 2017-08-02:
Okay guys, after researching a bit, I noticed that the data returned is in DMS and not in decimal degrees. Anyways, the coordinates still not being exactly and there's still a lot of error (1 kilometer more or less of error).

DexterInd, 2017-08-02:
A raw reading of the GPS string will give you DMS; [in the readme for the GPS section](https://github.com/DexterInd/GrovePi/blob/master/Software/Python/grove_gps/README.md) it is explained:  

> gps.latitude and gps.longitude are calculated to give you a Google Map appropriate format and make use of negative numbers to indicate either South or West

Are you using the example (~/GrovePi/Software/Python/grove_gps $ python GroveGPS.py) as it is written?

AlexRex, 2017-08-02:
Yes, I am using it as it was written. I tried also the script `dextergps.py` but I am not getting any output with that one.

DexterInd, 2017-08-04:
[Why not use the example found here](https://github.com/DexterInd/GrovePi/blob/master/Software/Python/grove_gps/dextergps.py), and the code commands shown:

```
	gps = GROVEGPS()
	while True:
		time.sleep(1)
		in_data = gps.read()
		if in_data != []:
			print in_data
      print gps.lat, gps.NS, gps.lon,gps.EW
      print gps.latitude, gps.longitude
```

This last command `print gps.latitude, gps.longitude` should print them in Google Maps ready coordinates.

AlexRex, 2017-08-04:
I already tried it and I am not getting any output, I had it working for 5 minutes without any output.

AlexRex, 2017-08-04:
I have two more GPS sensors but I don't want to open them without assuring they will work... Could it be related to the sensor itself?

DexterInd, 2017-08-04:
"Could it be related to the sensor itself" - I think we've been around this issue a few times; the sensor seems to be working fine.  The first issue was that you weren't in view of satellites, and the second issue was that you weren't converting data to the correct coordinates, right?  

I'm a little lost about where we are right now.  Referring to the latest post about it working for 5 minutes without any output, did you acquire GPS signals from satellites?

AlexRex, 2017-08-04:
Yes, if I try with the `GroveGPS.py` I am getting output with at least 10 satellites in view. With the `dextergps.py` I am not getting any output.

The problem with `GroveGPS.py` is that as far as I am getting data from the GPS, the data is not correct, getting output with an error of 1 kilometer or more. I checked my coordinates with some online tools to see if the were correct.

RobertLucian, 2017-08-04:
@AlexRex In my tests, I also saw an error of 100 meters up to 1 kilometer, so this may be the `GPS'` precision. All, after all, we're using `Seeed`'s sensor, so may be they should better know.

Here's what I think:

* Your `GPS` module is fine - it has no problems.
* The script you're running is also fine.
* I'm suspecting there are interferences between the `GPS` module and the `GrovePi` which leads me to the next bullet point.
* You said you're using a generic charger rated at `5V/2000mAh` - are you sure it's providing the right voltage at all times? If the voltage drops a lot for a very short period of time, the data exchanged with the `GPS` module can get corrupted. This way the data is going to appear as if it's written in a different encoding - this can justify the diamond-shaped question marks you're getting. Can you power the `GrovePi`/`Raspberry Pi` duo with an extra power supply? Or at least change your current one and power the system from a `USB 3.0` for instance. That's how I've done it and I didn't have issues.
Also, don't forget the `GPS` draws a couple hundreds of milliamps, so the module is quite power hungry.
* Check the wiring between the `GrovePi` and the `GPS` module. I also feel like the `Grove` cable is not completely plugged in.
* If you do have anything else attached to your `GrovePi`, please tell us about it and remove it.

Waiting to see what you come with now.

CleoQc, 2017-08-04:
I haven't touched the GPS module in over a year. When I wrote DexterGPS, I made it quite restrictive. It validates every incoming signal and if it's not perfectly formed, it rejects it.  Therefore if you're not getting readings with DexterGPS, then your module is mangling the data.  As @DexterInd  said, it could be a power issue.
It could also be interference from your environment. You may be seeing up to 10 satellites, but their signals are getting warped (and thus DexterGPS will reject them)

AlexRex, 2017-08-05:
@RobertLucian @CleoQc @DexterInd You were right, it was a power issue. I tried with a powerful power supply (an iPad one) and now it's working perfectly, also with `dexterind.py` script, and I am getting almost no error with the output coordinates. 

Thank you very much guys!

closed 2017-08-05 by AlexRex

DexterInd, 2017-08-07:
Great to hear!  thanks for letting us know.  I would _strongly_ recommend powering it off a dedicated source.  Typically a computer port only puts out 500 mA.  You'll need at LEAST 1 A to power the Pi alone, let alone the GrovePi shield or sensors.

AlexRex, 2017-08-07:
Yeah, I thought my power supply was strong enough. How could I power the grovepi or sensors with a secondary power supply? Is that even possible? 

Thanks!