DexterInd/GrovePi #365: Debugging code left in recent firmware versions

ndelC, 2017-07-28:
There are several places in recent firmware versions where there is debugging/testing code for the Grove dust sensor. My line numbers are in 1.2.7 as it is the most recent.

There is an "attachInterrupt" called on line 70 that gets called on startup. This is only supposed to be called when the Dust Sensor enable command (14) is sent:
```
void setup()
{
    // Serial.begin(38400);         // start serial for output
    Wire.begin(SLAVE_ADDRESS);

    Wire.onReceive(receiveData);
    Wire.onRequest(sendData);
	attachInterrupt(0,readPulseDust,CHANGE);
}
```
There are also two "digitalWrite" calls on lines 658 and 665. These shouldn't be here at all, they would make port D8 unusable when the Dust Sensor is used.
```
void readPulseDust()
{
  t = millis();
  l_status = digitalRead(2);  // Represents if the line is low or high.  
  if(l_status)
  { 
	 digitalWrite(8,0);
    // If the line is high (1), the pulse just ended
    pulse_end = t;
  }
  else
  {   // If the line is low (0), the pulse just started
    pulse_start = t;
	digitalWrite(8,1);
  }
  
  if(pulse_end > pulse_start)
  {
    duration = pulse_end - pulse_start;
    lowpulseoccupancy = lowpulseoccupancy+duration;   // Add to the pulse length.
    pulse_end = 0;    // If you don't reset this, you'll keep adding the pulse length over and over.
  }
}
```
Found this by having a temperature sensor plugged into D2 and an LED plugged into D8. Polling the temperature sensor caused the LED to blink.

nikkoura, 2017-07-28:
I had noticed that there was an interrupt attachment in the initialization code, while using myself interrupts (see PR #341). I am not sure that this is debug code, though. Rather just something coded that way, and which did not bothered anyone before :)

Also this hardcoded interrupt attachment forces the use of a specific grove port, whereas there is a second possible interrupt / port pair.

RobertLucian, 2017-07-29:
This seems important. 
As I already have a `Grove Dust Sensor`, I think I can test it the following week and come up with a diagnostic.

Thank you guys for letting me know.

RobertLucian, 2018-07-05:
The issue @ndelC has brought up continues in #408.

closed 2018-07-05 by RobertLucian