DexterInd/GrovePi #389: Can't get GPS module working

SYN-Vaibhav-Ranbhare, 2018-02-24:
I have a GPS module installed in a GrovePI, using the RPISER port. (Raspberry Pi 3).
 
`using ser = serial.Serial('dev/ttyAMA0',9600, timeout=0 ) # open serial port. 
... 
GPS.inp = ser.readline()
print(GPS.inp)
`
Result 
b''
b''

Using following config setting. 
dtoverlay=pi3-diable-bt
enable_uart=1
core_freq=250

nosarious, 2018-07-22:
I am having this problem as well in July. A solution would be nice.

CleoQc, 2018-08-20:
On a Pi3, or Pi3B+ some configuration is needed.

Step 1:
in /boot/config.txt, add at the end of the file:
```dtoverlay=pi3-miniuart-bt
dtoverlay=pi3-disable-bt
enable_uart=1```

Step 2:
in /boot/cmdline.txt  remove :
```console=ttyAMA0,115200 console=tty1```
and add:
```plymouth.ignore-serial-consoles```

Step 3:
reboot.