DexterInd/GrovePi #396: Issue installing node-grovepi

lucbors, 2018-04-04:
Hi,
I've been using GrovePi with raspberry pi for a couple of years now, and it works all fine for me.
Now I am preparing my Raspberry in order to rewrite all my stuff in Javascript. 
I just upgraded the node (to version 8.11.1) on my raspberry pi ( Pi2 Model B) My next step was installing node-grovepi. Unfortunately that fails:

`npm install node-grovepi
request to https://registry.npmjs.org/node-grovepi failed , reason:  connect ENETUNREACH 151.101.36.162:443
` 
Not sure where/what exactly goes wrong but it looks like there is something wrong with the URL, or with port 443 access...

Any help is appreciated.

Thanks

Luc Bors

marcellobarile, 2018-04-05:
Hello @lucbors,
could you please check if you are browsing behind a proxy?

Best,
Marcello

lucbors, 2018-04-06:
Hi Marcello,
I just checked it on a different network (office instead of customer) and it works just fine.
I will check the other network for proxy or IPv6. Consider the issue closed.
Thanks for your quick reply.

Luc

closed 2018-04-06 by lucbors

marcellobarile, 2018-04-06:
@lucbors, I'm glad that it worked.
If you are still facing such an issue please consider upgrading to a 9.x version of Node.js (NVM is a great tool for handling different versions, if you don't know it already).

Best,
Marcello