DexterInd/GrovePi #409: fix - I2C protocol issues (OSError/IOError) possibly caused by no support of repeated-starts on I2C

RobertLucian, 2018-07-05:
Someone also has reported issues when communicating with the dust sensor device when using the GrovePi. An I2C error is thrown, but it could very well be just a firmware issue - possibly there’s a mismatch between the GrovePi does and what the Pi expects. Also other IOError/OSError exceptions have been encountered along the way.

Personally, I haven't encountered this issue just yet.

Here's the list of posts where this was reported:
https://forum.dexterindustries.com/t/input-output-error-reading-dust-sensor/4466
https://forum.dexterindustries.com/t/grovepi-software-python-grovepi-py-line-227-in-analogread-return-number-1-256-number-2-typeerror-int-object-has-no-attribute-getitem/3412

ductape, 2019-01-07:
So I think the issue that I posted above may be related, but may not be. I had an issue in that the Seeed Grove Mini Motor Driver v1.1 uses default addresses of 0x60 and 0x65 whereas the grove_mini_motor_driver had addresses of 0x60 and 0x62. Even after fixing that mistake, there were still I2C issues with the `bus_i2c_block_read` when talking to the Arduino (address 0x04) which were fixed by updated to the currently unreleased v1.3.0 images in PR #423 . It looks like v1.3.0 release will close this issue and it fixed my problem.

RobertLucian, 2019-01-07:
> when talking to the Arduino (address 0x04) which were fixed by updated to the currently unreleased v1.3.0 images 

Are you actually referring to the GrovePi?

And yeah, the I2C issues with the GrovePi are fixed in #423. At the moment, I still have to fully test the GrovePi's beta firmware and then see if it works on the Pi Zero and only after that I'll be comfortable making it official.

ductape, 2019-01-08:
Yes, the combination of the firmware on the GrovePi and the Python software. Thanks for the update on the release progress!

closed 2019-04-27 by RobertLucian