DexterInd/GrovePi #417: digital_analog.py brings up error in grovepi.py

nosarious, 2018-07-22:
running the digital_analog.py script brings up the following error for line 227 in analogRead in grovepi.py:
   return[1] * 256 + number[2]
TypeError: 'int' object is not scriptable

this was accessed from line 57 of digital_analog.py:
   sensor_value0=grovepi.analogRead(sensor0)

as this is in the folder GrovePi/Software/Python/firmwaretests one would expect this to work

I cannot modify grovepi.py...

RobertLucian, 2018-07-23:
This looks like the GrovePi is not reachable. Have you tried flashing the firmware to it?

nosarious, 2018-07-24:
Flashing the firmware did make a difference, but with only two sensors (on 0 and 2) I get three values which seems... odd.

closed 2020-05-19 by CleoQc