DexterInd/GrovePi #418: temperature and humidity sensor does not read [solved]

yeeking, 2018-07-23:
I am testing the temp and humidity sensor that comes with the starter kit (the blue one).

The example code in the projects folder gives back nans. Here is the minimum code that produces the error, where the temp is printed once but not again:

from grovepi import *
from grove_oled import *

dht_sensor_port = 7		# Connect the DHt sensor to port 7

time.sleep(.1)

while True:
	try:
		[ temp,hum ] = dht(dht_sensor_port,1)		#Get the temperature and Humidity from the DHT sensor

		print("temp =", temp, "C\thumidity =", hum,"%") 
		time.sleep(1)	
	except (IOError,TypeError) as e:
		print("Error")

```
This is the output
```
pi@raspberrypi:~ $ python3 home_temp_hum_display.py 
temp = 28.0 C	humidity = 47.0 %
temp = nan C	humidity = nan %
temp = nan C	humidity = nan %

```

 However, I discovered that this does work:

```
from grovepi import *
from grove_oled import *

dht_sensor_port = 7		# Connect the DHt sensor to port 7

time.sleep(.1)

while True:
	try:
		dht(dht_sensor_port,0)
		time.sleep(0.5)
		[ temp,hum ] = dht(dht_sensor_port,1)		#Get the temperature and Humidity from the DHT sensor

		print("temp =", temp, "C\thumidity =", hum,"%") 
		time.sleep(1)	
	except (IOError,TypeError) as e:
		print("Error")
```

RobertLucian, 2018-07-23:
Thanks for the report. This looks like it's the exact same problem as in #411. I'm already debugging it and I'm trying to see where this comes from.

RobertLucian, 2018-07-27:
@yeeking in the first case, were NaNs the only thing you were seeing? If not, have you considered reading the DHT sensor until you'd get valid data and then repeat the process? 

What I have noticed is that the DHT sensor cannot be read too fast. When that happens, you get NaNs as a response - it's in the library of the sensor actually. So that's why it would be a probably good idea to run a loop with a sleep in it until you get valid data.

closed 2019-04-27 by RobertLucian