DexterInd/GrovePi #419: Cannot find module 'i2c-bus' even we have npm install i2c-bus

vkjuju, 2018-07-24:
We hit a weird error(Error: Cannot find module 'i2c-bus'
) even we have done" npm install i2c-bus", we have also checked:
$npm list|grep i2c-bus
├─┬ i2c-bus@3.2.0
│ ├─┬ i2c-bus@1.2.5
│ ├── i2c-bus@3.2.0 deduped
any advice would be appreciated.

Error: Cannot find module 'i2c-bus'
at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
at Function.Module._load (internal/modules/cjs/loader.js:507:25)
at Module.require (internal/modules/cjs/loader.js:637:17)
at require (internal/modules/cjs/helpers.js:20:18)
at Object. (/home/pi/Dexter/GrovePi/Software/NodeJS/libs/grovepi.js:2:19)
at Module._compile (internal/modules/cjs/loader.js:689:30)
at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
at Module.load (internal/modules/cjs/loader.js:599:32)
at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
at Function.Module._load (internal/modules/cjs/loader.js:530:3)
at Module.require (internal/modules/cjs/loader.js:637:17)
at require (internal/modules/cjs/helpers.js:20:18)
at Object. (/home/pi/Dexter/GrovePi/Software/NodeJS/libs/sensors/base/sensor.js:3:20)
at Module._compile (internal/modules/cjs/loader.js:689:30)
at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
at Module.load (internal/modules/cjs/loader.js:599:32)

RobertLucian, 2018-07-27:
I've just tried the NodeJS library and I haven't had any issues with it. The installation of `node-grovepi` went fine and that was it. And I already have done it a couple of times so it works.

And it doesn't make any sense for npm to not be able to install `i2c-bus` since that's already up on npm. Maybe you can uninstall these I2C packages, or better yet all of them and start from scratch again by running the install command. 

Just so you know, I've been doing this on my Stretch Raspbian For Robots image dated this summer (the latest one). These following details might also help you: the npm version is `5.6.0`,  node version is `9.11.2` and this is the package tree I get when I type `npm list`:

[image]

vkjuju, 2018-07-27:
please take a look at the following npmlist i2c-bus, does that mean my ic2-bus installed successfully ?

Thanks
Joe

npm list i2c-bus
rapiro-iotf@1.0.0 /home/pi/workspace/rapiro-iotf
├── i2c-bus@3.2.0
├─┬ node-grovepi@2.2.2
│ └── i2c-bus@1.2.5
└─┬ raspi-i2c@6.1.2
  └── i2c-bus@3.2.0  deduped

RobertLucian, 2018-07-27:
So it looks yeah. Was the re-installation that did the job for you? Are you now able to import the module in the console?

vkjuju, 2018-07-27:
no luck after reinstall , I don't know how to debug this issue, any more advice ? other modules running perfectly on my pi except this i2c-bus, I have been hanging there for over one week :(

vkjuju, 2018-07-28:
my node.js code is as follows, looks like serialport calling j2c-bus module:

const SerialPort = require('serialport');
const Readline = SerialPort.parsers.Readline;
const port = new SerialPort("/dev/ttys0",{
    baudRate: 57600,
    parser: new Readline("\n")
});

vkjuju, 2018-07-28:
but we hit another error:

pi@raspberrypi:~/workspace/rapiro-iotf $ sudo node robot_main.js
starting
info GrovePi.board GrovePi is initing
/home/pi/workspace/rapiro-iotf/robot_main.js:165
        soundSensor = new SoundAnalogSensor(2)
                      ^

TypeError: SoundAnalogSensor is not a constructor
    at onInit (/home/pi/workspace/rapiro-iotf/robot_main.js:165:16)
    at GrovePi.init (/home/pi/workspace/rapiro-iotf/node_modules/node-grovepi/grovepi.js:73:9)
    at start (/home/pi/workspace/rapiro-iotf/robot_main.js:268:9)
    at Object.<anonymous> (/home/pi/workspace/rapiro-iotf/robot_main.js:315:1)
    at Module._compile (internal/modules/cjs/loader.js:654:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:665:10)
    at Module.load (internal/modules/cjs/loader.js:566:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:506:12)
    at Function.Module._load (internal/modules/cjs/loader.js:498:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:695:10)
    at startup (internal/bootstrap/node.js:201:19)
    at bootstrapNodeJSCore (internal/bootstrap/node.js:516:3)

vkjuju, 2018-07-28:
node.js code:
function start() {
  console.log('starting')

  board = new Board({
    debug: true,
    onError: function(err) {
      console.log('TEST ERROR')
      console.log(err)
    },
    onInit: function(res) {
      if (res) {
        ultrasonicSensor = new UltrasonicDigitalSensor(2)
        lightSensor = new LightAnalogSensor(1)
        soundSensor = new SoundAnalogSensor(2)
        var buttonSensor = new ButtonDigitalSensor(5)

                 buttonSensor.stream(100,function(res){
                 if(res==1){
                 processing_music.stop();
                trigger=false;
                can_record=true;
                speaker_using=false;
                 sp.write("#M00\n");

                 }
                 }
                );

vkjuju, 2018-07-28:
I have opened a new issue:  https://forum.dexterindustries.com/t/typeerror-soundanalogsensor-is-not-a-constructor/5074