DexterInd/GrovePi #421: Flowmeter and Encoder both use port 2

ronski73, 2018-07-27:
I'm trying to use both the flowmeter and the encoder in one project. They work, but they both use D2 on the GrovePi+. Is there any way to make these modules work on other inputs? Currently, these numbers are hardcoded in the firmware (see grove_pi_v1_2_7.ino and Encoder.cpp).

Building a new firmware seems the only option.

RobertLucian, 2018-07-27:
Okay, that's interesting to me. I don't have a flowmeter and an encoder to test with, but I'm working on a couple of other fixes for the GrovePi. Theoretically, these other fixes we're working on could fix these 2 sensors you're talking about, but it's not a sure thing.

Would you be able to test the fixes once we have something?

RobertLucian, 2018-07-27:
This functionality was now added for the flow meter and the dust sensor. The same improvement hasn't been added to the encoder sensor because the library is badly written and trying to modify too much of it is risky - that's without any sensor to test with might.

Even in this situation, any combination of these 3 sensors in pairs of 2 can be used simultaneously on the GrovePi. Only 2 of them can be connected concurrently because there are only 2 external interrupts on the GrovePi: on port 2 and 3.

@ronski73 we will have something official as soon as we're done with our testing/approval.

closed 2019-04-27 by RobertLucian