DexterInd/GrovePi #424: Install grovepi without any graphical package

zlbabe, 2018-08-03:
Hi,
Is there any simplified way to install Grovepi without all graphical packages that come as dependencies ?
I'm running Raspbian on Raspberry pi 3 and when I try to install Grovepi it installs all these packages:
```
The following NEW packages will be installed:
  adwaita-icon-theme at-spi2-core dconf-gsettings-backend dconf-service fontconfig fontconfig-config fonts-dejavu-core glib-networking glib-networking-common glib-networking-services
  gnome-icon-theme gsettings-desktop-schemas gtk-update-icon-cache hicolor-icon-theme libatk-bridge2.0-0 libatk1.0-0 libatk1.0-data libatspi2.0-0 libavahi-client3 libavahi-common-data
  libavahi-common3 libcairo-gobject2 libcairo2 libcolord2 libcroco3 libcups2 libdatrie1 libdconf1 libdrm-amdgpu1 libdrm-freedreno1 libdrm-nouveau2 libdrm-radeon1 libepoxy0 libfontconfig1
  libgail-common libgail18 libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-common libgl1-mesa-dri libgl1-mesa-glx libglapi-mesa libglu1-mesa libgraphite2-3 libgtk-3-0 libgtk-3-bin libgtk-3-common
  libgtk2.0-0 libgtk2.0-bin libgtk2.0-common libharfbuzz0b libice6 libjbig0 libjpeg8 libjson-glib-1.0-0 libjson-glib-1.0-common liblcms2-2 libllvm3.9 libnotify4 libpango-1.0-0
  libpangocairo-1.0-0 libpangoft2-1.0-0 libpixman-1-0 libpng12-0 libproxy1v5 librest-0.7-0 librsvg2-2 librsvg2-common libsensors4 libsm6 libsoup-gnome2.4-1 libsoup2.4-1 libthai-data
  libthai0 libtiff5 libtxc-dxtn-s2tc libwxbase2.8-0 libwxbase3.0-0v5 libwxgtk2.8-0 libwxgtk3.0-0v5 libx11-xcb1 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-render0
  libxcb-shm0 libxcb-sync1 libxcomposite1 libxcursor1 libxdamage1 libxfixes3 libxi6 libxinerama1 libxrandr2 libxrender1 libxshmfence1 libxtst6 libxxf86vm1 notification-daemon python-psutil
  python-wxgtk2.8 python-wxgtk3.0 python-wxtools python-wxversion wx2.8-i18n x11-common
```
I want to stick with minimalist installation without any graphical package. 

Thanks,

RobertLucian, 2018-08-03:
I think you might be able to do this. Here, please take a look at this README for the install script of the GrovePi:
https://github.com/DexterInd/GrovePi/tree/master/Script

See that there are options that you can append to the install command. One particular option that's set to false by default is `--bypass-rfrtools`, which means the RFR Tools library is added. Considering that RFR Tools contains all the GUI things including the apt-get installs, you can just append this option to the install command:
```bash
curl -kL dexterindustries.com/update_grovepi | bash -s --bypass-rfrtools
```
Let me know if this worked out for you.

zlbabe, 2018-08-03:
unfortunately it doesn't
```
pi@raspberrypi:~ $ curl -kL dexterindustries.com/update_grovepi | bash -s -- --bypass-rfrtools
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   178  100   178    0     0    644      0 --:--:-- --:--:-- --:--:--   647
100   178  100   178    0     0    323      0 --:--:-- --:--:-- --:--:--  4139
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
100 11507  100 11507    0     0   7986      0  0:00:01  0:00:01 --:--:--  7986
Welcome to GrovePi Installer.
Updating GrovePi for master branch with the following options:
  --no-dependencies=false
  --no-update-aptget=false
  --bypass-rfrtools=true
  --bypass-python-rfrtools=false
  --user-local=false
  --env-local=false
  --system-wide=true
Using "master" branch
Options used for RFR_Tools script: "--system-wide master --use-python3-exe-too --update-aptget --install-deb-deps --install-python-package"
Options used for script_tools script: "master"
Installing script_tools. This might take a while..
Done installing script_tools
Installing GrovePi dependencies. This might take a while..
Special thanks to Joe Sanford at Tufts University. This script was derived from his work. Thank you Joe!
Installing dependencies for the GrovePi
Reading package lists... Done
Building dependency tree       
Reading state information... Done
nodejs is already the newest version (8.11.1~dfsg-2~bpo9+1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.11.0-3+deb9u3).
i2c-tools is already the newest version (3.1.2-3).
libi2c-dev is already the newest version (3.1.2-3).
The following additional packages will be installed:
  adwaita-icon-theme arduino-core at-spi2-core avr-libc avrdude binutils-avr ca-certificates-java dconf-gsettings-backend dconf-service default-java-plugin default-jre default-jre-headless
  extra-xdg-menus fontconfig fontconfig-config fonts-dejavu-core fonts-dejavu-extra gcc-avr glib-networking glib-networking-common glib-networking-services gsettings-desktop-schemas
  gtk-update-icon-cache hicolor-icon-theme icedtea-8-plugin icedtea-netx icedtea-netx-common java-common libasyncns0 libatk-bridge2.0-0 libatk-wrapper-java libatk-wrapper-java-jni
  libatk1.0-0 libatk1.0-data libatspi2.0-0 libavahi-client3 libavahi-common-data libavahi-common3 libcairo-gobject2 libcairo2 libcolord2 libcroco3 libcups2 libdatrie1 libdconf1
  libdrm-amdgpu1 libdrm-freedreno1 libdrm-nouveau2 libdrm-radeon1 libepoxy0 libflac8 libfontconfig1 libfontenc1 libftdi1 libgail-common libgail18 libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-common
  libgif7 libgl1-mesa-dri libgl1-mesa-glx libglapi-mesa libgraphite2-3 libgtk-3-0 libgtk-3-bin libgtk-3-common libgtk2.0-0 libgtk2.0-bin libgtk2.0-common libharfbuzz0b libice6 libjbig0
  libjna-java libjna-jni libjson-glib-1.0-0 libjson-glib-1.0-common liblcms2-2 libllvm3.9 libnspr4 libnss3 libogg0 libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libpixman-1-0
  libproxy1v5 libpulse0 librest-0.7-0 librsvg2-2 librsvg2-common librxtx-java libsensors4 libsm6 libsndfile1 libsoup-gnome2.4-1 libsoup2.4-1 libthai-data libthai0 libtiff5 libtxc-dxtn-s2tc
  libvorbis0a libvorbisenc2 libx11-xcb1 libxaw7 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-render0 libxcb-shape0 libxcb-shm0 libxcb-sync1 libxcomposite1 libxcursor1
  libxdamage1 libxfixes3 libxft2 libxi6 libxinerama1 libxmu6 libxpm4 libxrandr2 libxrender1 libxshmfence1 libxt6 libxtst6 libxv1 libxxf86dga1 libxxf86vm1 lrzsz openjdk-8-jre
  openjdk-8-jre-headless x11-common x11-utils
Suggested packages:
  arduino-mk avrdude-doc task-c-devel gcc-doc colord cups-common gvfs libjna-java-doc liblcms2-utils pulseaudio librsvg2-bin lm-sensors libnss-mdns fonts-ipafont-gothic
  fonts-ipafont-mincho fonts-wqy-microhei fonts-wqy-zenhei fonts-indic mesa-utils
The following NEW packages will be installed:
  adwaita-icon-theme arduino arduino-core at-spi2-core avr-libc avrdude binutils-avr ca-certificates-java dconf-gsettings-backend dconf-service default-java-plugin default-jre
  default-jre-headless extra-xdg-menus fontconfig fontconfig-config fonts-dejavu-core fonts-dejavu-extra gcc-avr glib-networking glib-networking-common glib-networking-services
  gsettings-desktop-schemas gtk-update-icon-cache hicolor-icon-theme icedtea-8-plugin icedtea-netx icedtea-netx-common java-common libasyncns0 libatk-bridge2.0-0 libatk-wrapper-java
  libatk-wrapper-java-jni libatk1.0-0 libatk1.0-data libatspi2.0-0 libavahi-client3 libavahi-common-data libavahi-common3 libcairo-gobject2 libcairo2 libcolord2 libcroco3 libcups2
  libdatrie1 libdconf1 libdrm-amdgpu1 libdrm-freedreno1 libdrm-nouveau2 libdrm-radeon1 libepoxy0 libflac8 libfontconfig1 libfontenc1 libftdi1 libgail-common libgail18 libgdk-pixbuf2.0-0
  libgdk-pixbuf2.0-common libgif7 libgl1-mesa-dri libgl1-mesa-glx libglapi-mesa libgraphite2-3 libgtk-3-0 libgtk-3-bin libgtk-3-common libgtk2.0-0 libgtk2.0-bin libgtk2.0-common
  libharfbuzz0b libice6 libjbig0 libjna-java libjna-jni libjson-glib-1.0-0 libjson-glib-1.0-common liblcms2-2 libllvm3.9 libnspr4 libnss3 libogg0 libpango-1.0-0 libpangocairo-1.0-0
[... cut ...]
  libthai0 libtiff5 libtxc-dxtn-s2tc libvorbis0a libvorbisenc2 libx11-xcb1 libxaw7 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-render0 libxcb-shape0 libxcb-shm0
  libxcb-sync1 libxcomposite1 libxcursor1 libxdamage1 libxfixes3 libxft2 libxi6 libxinerama1 libxmu6 libxpm4 libxrandr2 libxrender1 libxshmfence1 libxt6 libxtst6 libxv1 libxxf86dga1
  libxxf86vm1 lrzsz minicom openjdk-8-jre openjdk-8-jre-headless x11-common x11-utils
0 upgraded, 136 newly installed, 0 to remove and 0 not upgraded.
Need to get 79.2 MB/100 MB of archives.
After this operation, 515 MB of additional disk space will be used.
```

RobertLucian, 2018-08-07:
@zlbabe we have referenced #425 which can help you bypass the installation of GUI packages. This will reduce the amount of space that's consumed. 

On a basic Stretch Lite image if I bypass these I only need to download 72 megabytes of archives and the installation size of these debian packages goes to 120 megabytes. And that's containing the I2C tools, git, python(3)-setuptools, etc that's needed. 

Hopefully, we should roll it out in not much time.