DexterInd/GrovePi #435: LCD RGB screen not working

vnahmias, 2018-09-27:
Hello, here is my problem : i call 

```
def alertHuman(self,msg):
        setText_norefresh("Light : " + str(msg))
```
in a while loop and after like 10 iterations it crashes and i have this error message : 
```
File "/usr/lib/python3.5/multiprocessing/process.py", line 249, in _bootstrap
    self.run()
  File "/usr/lib/python3.5/multiprocessing/process.py", line 93, in run
    self._target(*self._args, **self._kwargs)
  File "component1.py", line 55, in runHuis
    LCD.alertHuman(value)
  File "/home/pi/grovePiDataSender/IoT-DataSender/Entities/HumanUserInterfaces/GroveHUIs/GroveLCD.py", line 12, in alertHuman
    setText_norefresh("Light : " + str(msg))
  File "/home/pi/grovePiDataSender/IoT-DataSender/Entities/HumanUserInterfaces/GroveHUIs/grove_rgb_lcd.py", line 77, in setText_norefresh
    textCommand(0x02) # return home
  File "/home/pi/grovePiDataSender/IoT-DataSender/Entities/HumanUserInterfaces/GroveHUIs/grove_rgb_lcd.py", line 52, in textCommand
    bus.write_byte_data(DISPLAY_TEXT_ADDR,0x80,cmd)
  File "/usr/local/lib/python3.5/dist-packages/smbus_cffi-0.5.1-py3.5-linux-armv7l.egg/smbus/util.py", line 59, in validator
    return fn(*args, **kwdefaults)
  File "/usr/local/lib/python3.5/dist-packages/smbus_cffi-0.5.1-py3.5-linux-armv7l.egg/smbus/smbus.py", line 145, in write_byte_data
    raise IOError(ffi.errno)
```
What's happening here because i have not a single clue

eternasparta, 2018-11-27:
Do you have a solution?

CleoQc, 2018-11-27:
Apologies, this fell between the cracks. Can you post your code?
From what you mentioned, it seems you would need to add a sleep command in your loop so that it doesn't look too quickly.  Something like `sleep(0.5)`  or even `sleep(0.2)`  but I cannot say for sure until we see your code.

jasondeutsch, 2020-09-20:
Any update on this?