DexterInd/GrovePi #436: Error using high temperature sensor

benjaminforest, 2019-01-07:
here is my callstack : 
```
Traceback (most recent call last):
  File "high_temperature_example.py", line 66, in <module>
    Main()
  File "high_temperature_example.py", line 54, in Main
    room_temperature = sensor.getRoomTemperature()
  File "/home/pi/GrovePi/Software/Python/grove_hightemperature_sensor/grove_hightemperature_sensor.py", line 61, in getRoomTemperature
    analog_sum += grovepi.analogRead(self.temperature_pin)
  File "/usr/local/lib/python3.5/dist-packages/grovepi-1.0.4-py3.5.egg/grovepi.py", line 227, in analogRead
TypeError: 'int' object is not subscriptable
```

I did follow the folowing procedure : https://www.dexterindustries.com/GrovePi/get-started-with-the-grovepi/setting-software/, and everything worked fine.

Any tips ?

benjaminforest, 2019-01-07:
Note that analog read do not work either : 
```
python3 grove_analog_read.py
Traceback (most recent call last):
  File "grove_analog_read.py", line 44, in <module>
    sensor_value = grovepi.analogRead(sensor)
  File "/home/pi/GrovePi/Software/Python/grovepi.py", line 227, in analogRead
    return number[1] * 256 + number[2]
TypeError: 'int' object is not subscriptable`
```

RobertLucian, 2019-01-07:
Check the firmware version by running `grovepi.version()`. Possibly, even try to update the firmware (not the library on the, but the firmware). 

The old 1.2.7 firmware (which is still the official one) is riddled with bugs and that's why we're pushing for #423 . Getting that `int object is not subscriptable` error is common for this version.

benjaminforest, 2019-01-08:
Thanks for this quick answer. Unfortunately, I didn't managed to update the firmware : 
```
Checking for Atmega chip
========================

avrdude: Version 5.10, compiled on Jun 18 2012 at 12:38:29
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : unknown
         Using Programmer              : gpio
         AVR Part                      : ATMEGA328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     5     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : GPIO
         Description     : Use sysfs interface to bitbang GPIO lines

avrdude: AVR device not responding
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.
```
I'm looking for a solution, without much success for the moment (the i2c part seems to be working).

benjaminforest, 2019-01-08:
Note : I tried Raspbian for robots as well, without more success

benjaminforest, 2019-01-08:
Note 2 : I bought my board from seeedstudio, who suggest to use dexter firmware. Maybe it is linked ?

RobertLucian, 2019-01-08:
It could be a broken trace on the SPI line. Did you make sure you had the SPI interface enabled? There's a way to diagnose that and if you have the time and patience you can follow this comment here:
https://forum.dexterindustries.com/t/solved-grovepi-firmware-update-fails-avr-device-not-responding/3144/8

> who suggest to use dexter firmware. Maybe it is linked ?

What do you mean by using the Dexter firmware. It sounds as if you're implying there are other firmwares too. Anyway, just to be sure about this, the only firmware that we know of and support is the one from this repo.

Also, you didn't tell me what is the output of running `grovepi.version()` instruction. Can you tell me what it is? Does it even work to run it?

benjaminforest, 2019-01-08:
Thanks a lot. After some more investigation I understand that I do have a grove **base** hat for pi and not a grovepi+ hat. This is a different microcontroller, no risk to flash it !
I was confused by seed https://www.seeedstudio.com/Grove-Base-Hat-for-Raspberry-Pi-p-3186.html
Thanks again.

closed 2019-01-08 by benjaminforest