DexterInd/GrovePi #438: Cannot get started with node.js sample

wmmihaa, 2019-01-29:
I have a newly installed pi3 with node.js 9.9.0 and npm 5.6.0. I created a groove directory and installed the node-grovepi package using:
```npm install node-grovepi```

I also ran the quick install command:
```curl -kL dexterindustries.com/update_grovepi | bash```

It seems to all be installed correctly, but I don't quite  get it working. When trying the analog light for instance, I keep getting the same value (~20.0), but it makes no difference if I cover the sensor or hold it up to the light. If I put the sensor on a different port I get different readings (~0.01). So it seems like it can read the sensor but are not getting the right values.

If I try the leds nothing happends, not even an error...
``` javascript
var led = new GrovePi.sensors.DigitalOutput(2);
board = new GrovePi.board({...});
led.turnOn();
```
Appreciate any help ;)

wmmihaa, 2019-01-29:
Thanks @marcellobarile, 
That is actually what I do (copied from the sample). For some reason I choose hide it with three dots ;)

marcellobarile, 2019-01-29:
alright but then the "led.turnOn" call should be placed inside the "onInit" handler. 
You can find something more here: https://github.com/DexterInd/GrovePi/tree/master/Software/NodeJS/tests

wmmihaa, 2019-01-29:
Could there be any type of configurations of the pi I could have missed? I've enabled ic2 and gpio along with running ```curl -kL dexterindustries.com/update_grovepi | bash```

wmmihaa, 2019-01-30:
Ad I didn't use the designated sd card (lost it), is this image downloadable?

RobertLucian, 2019-01-30:
Well, you can download Raspbian For Robots that already has everything in it, or you can get a Raspbian and then run the above command and install it. Regardless, you have to run the same above command in order to get the GrovePi lib (mostly the python stuff) up to date. 

Here's the link to the image:
https://www.dexterindustries.com/howto/install-raspbian-for-robots-image-on-an-sd-card/

Also, personally, I'd go with a fresh distribution.