DexterInd/GrovePi #442: grovepi analogRead() stuck

SnoringBoy, 2019-03-02:
Hi,
  I am having an issue that when I call the method grovepi.analogRead(), it is taking forever just like an infinite loop.  For the grove_rotary_angle_sensor.py, I don't have any errors on the console.  When I test other python programs like grove_led_fade.py.  The grovepi module works.  I have set up the grovepi by following official website steps.  I am very sure the analogRead() brings the problem since I put print statements before and after the analogRead() call.  The one before method call is printed.  Does anyone know what is happing?  Thank you, your help is appreciated.

avnish19, 2019-03-17:
I'm having the same issue. 

When placing some simple debug statements in my code, any script I am running stops at a analogRead() call.

I am using GrovePi Zero in my case.

thisisandrei, 2019-04-01:
Same issue here, I am using GrovePi+ with Raspberry Pi Zero W, with Raspbian installed. Anyone figured this out yet?

thisisandrei, 2019-04-01:
So the issue, at least for me, was that I needed to update the firmware, else analogRead was hanging out of the box. Also, if using Raspbian, run all installation and setup commands with sudo. 

Specifically, go into the Firmware folder of this repo, chmod 777 *, sudo ./new_fw_search.sh, sudo firmware_update.sh.

I hope this helps. The installation details for getting this going on Raspbian isn't the greatest but it works if you know what you're supposed to do. The out of the box install obviously didn't do a firmware update, even though I followed all instructions.

RobertLucian, 2019-04-01:
As far as I remember, this is all covered in the getting started section of the GrovePi's documentation [here](https://dexterind.github.io/GrovePi/fw/).

Were you not aware of the documentation? If so, where would you suggest having it put? As of this moment, it's found somewhere in the middle on the repo's README. Others also had problems finding it.

 Și salutări ție! Ești al 3-lea român pe care îl văd pe aici.

closed 2019-07-05 by RobertLucian