DexterInd/GrovePi #456: installing on fresh Raspbian with raspberry 4

ctheprinter, 2019-08-07:
Just trying to install Grovepi+ for the second time on a Raspberry pi 4. coming up with errors on the `curl -kL dexterindustries.com/update_grovepi | bash` request.

will post as soon as it finishes again. previously i tried putting the grovepi on then running the simple led blink test and nothing happened, but no errors in running the python program.

any help would be appreciated. i might try updating the firmware on the grovepi after the curl finishes.

ctheprinter, 2019-08-07:
Installed /tmp/easy_install-SIfdQA/smbus-cffi-0.5.1/.eggs/pycparser-2.19-py2.7.egg
creating /usr/local/lib/python2.7/dist-packages/smbus_cffi-0.5.1-py2.7-linux-armv7l.egg
Extracting smbus_cffi-0.5.1-py2.7-linux-armv7l.egg to /usr/local/lib/python2.7/dist-packages
Adding smbus-cffi 0.5.1 to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/smbus_cffi-0.5.1-py2.7-linux-armv7l.egg
Searching for python-periphery
Reading https://pypi.org/simple/python-periphery/
Downloading https://files.pythonhosted.org/packages/52/0e/b5bf7e8cfee3fbb843c2f5b5399da9a65e7c985b67e18f6e73d6da4fb132/python_periphery-1.1.2-py2.py3-none-any.whl#sha256=40292acf5afedc67d820598aec2cfa18ab2ff6aaa016991d4e5547538f174fa3
Best match: python-periphery 1.1.2
Processing python_periphery-1.1.2-py2.py3-none-any.whl
Installing python_periphery-1.1.2-py2.py3-none-any.whl to /usr/local/lib/python2.7/dist-packages
Adding python-periphery 1.1.2 to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/python_periphery-1.1.2-py2.7.egg
Searching for cffi>=1.1.0
Reading https://pypi.org/simple/cffi/
Downloading https://files.pythonhosted.org/packages/93/1a/ab8c62b5838722f29f3daffcc8d4bd61844aa9b5f437341cc890ceee483b/cffi-1.12.3.tar.gz#sha256=041c81822e9f84b1d9c401182e174996f0bae9991f33725d059b771744290774
Best match: cffi 1.12.3
Processing cffi-1.12.3.tar.gz
Writing /tmp/easy_install-v7VaF5/cffi-1.12.3/setup.cfg
Running cffi-1.12.3/setup.py -q bdist_egg --dist-dir /tmp/easy_install-v7VaF5/cffi-1.12.3/egg-dist-tmp-cG4YaL
creating /usr/local/lib/python2.7/dist-packages/cffi-1.12.3-py2.7-linux-armv7l.egg
Extracting cffi-1.12.3-py2.7-linux-armv7l.egg to /usr/local/lib/python2.7/dist-packages
Adding cffi 1.12.3 to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/cffi-1.12.3-py2.7-linux-armv7l.egg
Searching for pycparser
Reading https://pypi.org/simple/pycparser/
Downloading https://files.pythonhosted.org/packages/68/9e/49196946aee219aead1290e00d1e7fdeab8567783e83e1b9ab5585e6206a/pycparser-2.19.tar.gz#sha256=a988718abfad80b6b157acce7bf130a30876d27603738ac39f140993246b25b3
Best match: pycparser 2.19
Processing pycparser-2.19.tar.gz
Writing /tmp/easy_install-nfNzzO/pycparser-2.19/setup.cfg
Running pycparser-2.19/setup.py -q bdist_egg --dist-dir /tmp/easy_install-nfNzzO/pycparser-2.19/egg-dist-tmp-G2IlMW
warning: no previously-included files found matching 'setup.pyc'
warning: no previously-included files matching 'yacctab.*' found under directory 'tests'
warning: no previously-included files matching 'lextab.*' found under directory 'tests'
warning: no previously-included files matching 'yacctab.*' found under directory 'examples'
warning: no previously-included files matching 'lextab.*' found under directory 'examples'
zip_safe flag not set; analyzing archive contents...
pycparser.ply.lex: module references __file__
pycparser.ply.lex: module MAY be using inspect.getsourcefile
pycparser.ply.yacc: module references __file__
pycparser.ply.yacc: module MAY be using inspect.getsourcefile
pycparser.ply.yacc: module MAY be using inspect.stack
pycparser.ply.ygen: module references __file__
creating /usr/local/lib/python2.7/dist-packages/pycparser-2.19-py2.7.egg
Extracting pycparser-2.19-py2.7.egg to /usr/local/lib/python2.7/dist-packages
Adding pycparser 2.19 to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/pycparser-2.19-py2.7.egg
Searching for pyserial==3.4
Best match: pyserial 3.4
Adding pyserial 3.4 to easy-install.pth file

Using /usr/lib/python2.7/dist-packages
Finished processing dependencies for Dexter-AutoDetection-and-I2C-Mutex==0.0.0
running install
running bdist_egg
running egg_info
writing Dexter_AutoDetection_and_I2C_Mutex.egg-info/PKG-INFO
writing dependency_links to Dexter_AutoDetection_and_I2C_Mutex.egg-info/dependency_links.txt
writing requirements to Dexter_AutoDetection_and_I2C_Mutex.egg-info/requires.txt
writing top-level names to Dexter_AutoDetection_and_I2C_Mutex.egg-info/top_level.txt
reading manifest file 'Dexter_AutoDetection_and_I2C_Mutex.egg-info/SOURCES.txt'
writing manifest file 'Dexter_AutoDetection_and_I2C_Mutex.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build/lib
[... 867 lines cut ...]
Best match: numpy 1.16.2
Adding numpy 1.16.2 to easy-install.pth file
Installing f2py script to /usr/local/bin
Installing f2py3 script to /usr/local/bin
Installing f2py3.7 script to /usr/local/bin

Using /usr/lib/python3/dist-packages
Searching for RPi.GPIO==0.7.0
Best match: RPi.GPIO 0.7.0
Adding RPi.GPIO 0.7.0 to easy-install.pth file

Using /usr/lib/python3/dist-packages
Searching for cffi==1.12.3
Best match: cffi 1.12.3
Processing cffi-1.12.3-py3.7-linux-armv7l.egg
cffi 1.12.3 is already the active version in easy-install.pth

Using /usr/local/lib/python3.7/dist-packages/cffi-1.12.3-py3.7-linux-armv7l.egg
Searching for pycparser==2.19
Best match: pycparser 2.19
Processing pycparser-2.19-py3.7.egg
pycparser 2.19 is already the active version in easy-install.pth

Using /usr/local/lib/python3.7/dist-packages/pycparser-2.19-py3.7.egg
Finished processing dependencies for grovepi==1.0.4

RobertLucian, 2019-08-07:
Okay. When you first install the GrovePi, you are also required to reboot the Pi. Only then you can start interfacing with it. (I haven't looked over your logs yet).

RobertLucian, 2019-08-07:
The error you are getting on wxgtk2.8 is normal on the Buster image. This package is required for the GUI interface which I think you don't need, which means you can bypass the installation of it by running:
```
curl -kL dexterindustries.com/update_grovepi | bash -s -- --bypass-gui-installation
```

Either way, there's a fix out there in the pipeline for wxgtk2.8, but we haven't merged it yet because we need to test it. 

Truth to be told, we don't support any of our robots (or devices) on Buster/Pi4 because this distribution is still very buggy and because we're still working on making everything work smoothly. It's in the pipeline, but it's gonna take a bit more time until it's ready.

ctheprinter, 2019-08-07:
log after running all_tests.sh

seems to get stuck at checking firmware

Check space left
================
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        26G  4.5G   21G  19% /
devtmpfs        1.8G     0  1.8G   0% /dev
tmpfs           1.9G     0  1.9G   0% /dev/shm
tmpfs           1.9G  8.6M  1.9G   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           1.9G     0  1.9G   0% /sys/fs/cgroup
/dev/mmcblk0p6  253M   40M  213M  16% /boot
tmpfs           386M  4.0K  386M   1% /run/user/1000

Check for dependencies
======================
python 2.7.16-1 install ok installed
python-pip 18.1-5+rpt1 install ok installed
git 1:2.20.1-2 install ok installed
libi2c-dev 4.1-1 install ok installed
python-serial 3.4-4 install ok installed
python-rpi.gpio 0.7.0~buster-1 install ok installed
i2c-tools 4.1-1 install ok installed
python-smbus 4.1-1 install ok installed
dpkg-query: no packages found matching arduino
dpkg-query: no packages found matching minicom
scratch 1.4.0.6~dfsg1-6 install ok installed

find: ‘/run/user/1000/gvfs’: Permission denied
find: ‘/proc/997/task/997/net’: Invalid argument
find: ‘/proc/997/net’: Invalid argument
find: ‘/proc/1264/task/1264/net’: Invalid argument
find: ‘/proc/1264/net’: Invalid argument
find: ‘/proc/2660/task/2660/net’: Invalid argument
find: ‘/proc/2660/net’: Invalid argument
find: ‘/proc/7059/task/7059/net’: Invalid argument
find: ‘/proc/7059/net’: Invalid argument
wiringPi Not Found (ERR)
find: ‘/run/user/1000/gvfs’: Permission denied
find: ‘/proc/997/task/997/net’: Invalid argument
find: ‘/proc/997/net’: Invalid argument
find: ‘/proc/1264/task/1264/net’: Invalid argument
find: ‘/proc/1264/net’: Invalid argument
find: ‘/proc/2660/task/2660/net’: Invalid argument
find: ‘/proc/2660/net’: Invalid argument
find: ‘/proc/7059/task/7059/net’: Invalid argument
find: ‘/proc/7059/net’: Invalid argument
I2C still in blacklist (ERR)
SPI still in blacklist (ERR)

Check for addition in /modules
==============================
I2C-dev already there
i2c-bcm2708 already there
spi-dev already there

Hardware revision
=================
gpio version: 2.50
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Unknown17, Revision: 01, Memory: 0MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.1
  * This Raspberry Pi supports user-level GPIO access.
[... 84 lines cut ...]
avrdude: safemode: Fuses OK

avrdude done.  Thank you.

Checking I2C bus for devices
============================

Checking I2C bus 0
==================
Error: Could not open file `/dev/i2c-0' or `/dev/i2c/0': No such file or directory

Checking I2C bus 1
==================
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- 04 -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

Checking for firmware version
=============================

CleoQc, 2019-08-07:
You need to upgrade the firmware on the GrovePi first. 

`sudo bash /home/pi/Dexter/GrovePi/Firmware/firmware_update.sh `  should fix that.

ctheprinter, 2019-08-08:
trying to run 
```
import grovepi
print(grovepi.version)
```
but i get nothing...

when running all tests, it still hangs on checking firmware version

=============================
GrovePi Troubleshooting Script
=============================

Adding permissions to the scripts
=================================

Check space left
================
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        26G  4.5G   21G  19% /
devtmpfs        1.8G     0  1.8G   0% /dev
tmpfs           1.9G     0  1.9G   0% /dev/shm
tmpfs           1.9G   17M  1.9G   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           1.9G     0  1.9G   0% /sys/fs/cgroup
/dev/mmcblk0p6  253M   40M  213M  16% /boot
tmpfs           386M  4.0K  386M   1% /run/user/1000

Check for dependencies
======================
python 2.7.16-1 install ok installed
python-pip 18.1-5+rpt1 install ok installed
git 1:2.20.1-2 install ok installed
libi2c-dev 4.1-1 install ok installed
python-serial 3.4-4 install ok installed
python-rpi.gpio 0.7.0~buster-1 install ok installed
i2c-tools 4.1-1 install ok installed
python-smbus 4.1-1 install ok installed
dpkg-query: no packages found matching arduino
dpkg-query: no packages found matching minicom
scratch 1.4.0.6~dfsg1-6 install ok installed

find: ‘/run/user/1000/gvfs’: Permission denied
find: ‘/proc/915/task/915/net’: Invalid argument
find: ‘/proc/915/net’: Invalid argument
wiringPi Not Found (ERR)
find: ‘/run/user/1000/gvfs’: Permission denied
find: ‘/proc/915/task/915/net’: Invalid argument
find: ‘/proc/915/net’: Invalid argument
I2C still in blacklist (ERR)
SPI still in blacklist (ERR)

Check for addition in /modules
==============================
I2C-dev already there
i2c-bcm2708 already there
spi-dev already there

Hardware revision
=================
gpio version: 2.50
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Unknown17, Revision: 01, Memory: 0MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.1
  * This Raspberry Pi supports user-level GPIO access.
[... 75 lines cut ...]
         this check.

avrdude done.  Thank you.

Checking I2C bus for devices
============================

Checking I2C bus 0
==================
Error: Could not open file `/dev/i2c-0' or `/dev/i2c/0': No such file or directory

Checking I2C bus 1
==================
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

Checking for firmware version
=============================

CleoQc, 2019-08-08:
Did you run the command to upgrade the firmware?

ctheprinter, 2019-08-08:
Yes

ctheprinter, 2019-08-12:
I’ve ordered a pre-installed SD card https://thepihut.com/products/dexter-8gb-microsd-card-with-raspbian-for-robots

So I’ll see how that works...

In the mean time any ideas on what is going wrong or what I’m doing wrong?

mattallen37, 2019-08-12:
As Robert mentioned, none of the Dexter Industries devices are supported (yet) on Buster or Raspberry Pi 4.

I'm looking into what's failing and how to fix it, but meanwhile I'd recommend using the latest release of Raspbian for Robots with a Raspberry Pi 3 (or B+, 2, or 3B+).

There isn't a released version of Raspbian for Robots that's based on Buster, so none of the released RfR images will boot on a RPi4.

RobertLucian, 2019-08-12:
In the meantime @ctheprinter, I think you'd be better off with switching the Raspberry Pi to use the HW I2C. Check out this page here:
https://dexterind.github.io/GrovePi/quickstart/#what-i2c-bus-to-use

So, in your situation, every time you run a script intended for the GrovePi, run this instruction too at the beginning of it:
```python
grovepi.set_bus("RPI_1")
```

Let us know if this makes it work for you.

ctheprinter, 2019-08-15:
no, still doest do anything, when i call grovepi.version() it just hangs and doesn't do anything after it. although doesn't give any errors either. although i'm no good with python.

johnnyrich0617, 2020-01-23:
Should this issue be closed now that support for Buster is complete?

closed 2020-02-27 by CleoQc