DexterInd/GrovePi #457: Setup Documentation needs to be updated or clarify

Lassesen, 2019-08-21:
As a semi-retired tech writer, I found that the instructions were deficient.

My Steps (which I would expect someone should be doing periodically to validate documentation):

1. up fresh PI 3
2. ran all of the updates for PI.
3. EXECUTED: curl -kL dexterindustries.com/update_grovepi | bash   FINE
4. Literally followed the next instruction
5. cd Firmware  FOLDER NOT FOUND -- DEFECT IN DOCUMENTATION IMHO
6. Located Firmware folder at /home/pi/Dexter/GrovePi , changed to that folder.
7. EXECUTE bash firmware_update.sh   FAILED libusb-0.1.so.4: cannot open..

To resolve the last item, I had to do

apt-get install libusb-0.1-4

Again, as a former MSDN Tech writer, the initial setup instructions should be 100% reliable and idiot proof.

gavan1, 2020-02-23:
On Pi 3 A+ kernel 4.19 my folder dir.

`/home/pi/Dexter/lib/Dexter`

is there another setup command required  for the GrovePi folder?

CleoQc, 2020-02-27:
Thank you for your feedback. I added the changes, except for libusb which I still have to investigate.

closed 2020-03-03 by CleoQc