DexterInd/GrovePi #460: Installation on user than "pi"

sir-rave-alot, 2019-09-09:
Hi,
i'd like to install the library on another user than "pi", which is not possible using the install scripts as they are. Is anyone able to offer a patch for that? I tried to replace all the paths with sed regex but in the end it did not work, so I did the workaround using pi user, but this is not satisfying, as i don't want a user just for this library.

Thank you very much, 
sir-rave-alot

CleoQc, 2019-09-10:
Hi,
for historical reasons it's getting quite hard to get rid of the pi user. However would it be okay in your case to install with this user and then get rid of the user ? The libraries are installed globally, and don't depend on the pi user. We've had others who successfully got rid of the default user after installation.

sir-rave-alot, 2019-09-10:
Thank you for your quick response,
i will keep the pi user on the platform for that reason.
Not very nice but on the other hand not very costy.

Best regards,
sir-rave-alot

closed 2019-09-10 by sir-rave-alot