DexterInd/GrovePi #469: GrovePi folder not created.

gavan1, 2020-02-23:
Pi 3 A+ kernel 4.19 

`curl -kL dexterindustries.com/update_grovepi | bash`

my folder dir.

`/home/pi/Dexter/lib/Dexter`

is there another setup command required  for the GrovePi folder and files?

CleoQc, 2020-02-27:
that would be the main command. 
Are you saying that the /home/pi/Dexter/GrovePi folder hasn't been created?

CleoQc, 2020-02-27:
Ok, I can repro this in-house. At the moment, I'm baffled. I'm going to do a deep-dive into the scripts.

CleoQc, 2020-02-27:
Hmm, curiouser and curiouser.
I have a workaround - which worked for me, but no explanations yet.
```
curl -kL dexterindustries.com/update_grovepi > grovepi_setup.sh
bash grovepi_setup.sh
```
it makes no sense at the moment that the oneliner would fail while this two liner approach works. 

@gavan1 , could you try it? Does it work for you?

CleoQc, 2020-03-06:
Fixed with https://github.com/DexterInd/RFR_Tools/pull/40

closed 2020-03-06 by CleoQc