DexterInd/GrovePi #472: Installing GrovePi: ./install.sh: command not found

jonathanrjpereira, 2020-03-03:
**Installation Steps Followed:**

```
git clone https://github.com/DexterInd/GrovePi.git
cd GrovePi/Script
sudo chmod +x install.sh
sudo ./install.sh

```

**Error:**

```
pi@raspberrypi:~/GrovePi/Script $ sudo ./install.sh 
./install.sh: line 11: /home/pi/Dexter/lib/Dexter/script_tools/functions_library.sh: No such file or directory
Special thanks to Joe Sanford at Tufts University. This script was derived from his work. Thank you Joe!
./install.sh: line 38: feedback: command not found
I2C-dev already present
i2c-bcm2708 already present
spi-dev already present
./install.sh: line 58: feedback: command not found
i2c1 already present
i2c_arm already present
adduser: The group `i2c' already exists.
./install.sh: line 76: feedback: command not found
./install.sh: line 77: /home/pi/Dexter/lib/Dexter/script_tools/install_avrdude.sh: No such file or directory
./install.sh: line 78: create_avrdude_folder: command not found
./install.sh: line 79: install_avrdude: command not found
done with AVRDUDE for the GrovePi
```
**Raspberry Pi Specs:**
Raspberry Pi 3B+
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

CleoQc, 2020-03-03:
Please follow the install command from the readme file:

`curl -kL dexterindustries.com/update_grovepi | bash`

It pulls it more than one repo.

jonathanrjpereira, 2020-03-05:
Solved!

So these are the steps, I followed on the Raspberry Pi 3B+ with Raspbian Buster.

In the terminal, I ran the following commands:

```
git clone https://github.com/DexterInd/GrovePi

curl -kL dexterindustries.com/update_grovepi | bash

cd /home/pi/Dexter/GrovePi/Firmware
bash firmware_update.sh
```

I didn't have to do any of the previous installation steps as mentioned in the above comment.

closed 2020-03-05 by jonathanrjpereira

erikvandorp, 2020-03-22:
hey Guys, I have the same issue, but the solution that worked for Jonathan did not work for me. I did get a few steps further than before, but now I am stuck here:

I typed in : 
pi@raspberrypi:~/dexter/GrovePi/Firmware $ bash firmware_update.sh

Terminal said:
Updating the GrovePi firmware
=============================
 http://www.dexterindustries.com/grovepi 
 Run this program: 
 sudo ./firmware_update.sh
 
=============================
Do you want to update the firmware? [y,n]y
Make sure that GrovePi is connected to Raspberry Pi
Firmware found
Press any key to start firmware update
. . .
/home/pi/dexter/GrovePi/Firmware/grovepi_firmware_update.sh: line 3: avrdude: command not found
/home/pi/dexter/GrovePi/Firmware/grovepi_firmware_update.sh: line 4: avrdude: command not found
/home/pi/dexter/GrovePi/Firmware/grovepi_firmware_update.sh: line 5: avrdude: command not found
/home/pi/dexter/GrovePi/Firmware/grovepi_firmware_update.sh: line 6: avrdude: command not found

jonathanrjpereira, 2020-03-22:
I tried to reinstall Grove Pi on a fresh install of Raspbian and I came across a similar issue as shown in [https://github.com/DexterInd/GrovePi/issues/477](https://github.com/DexterInd/GrovePi/issues/477)