DexterInd/GrovePi #477: No such file or directory -- bash firmware_update.sh

jonathanrjpereira, 2020-03-13:
**I followed the following installation steps on a fresh installation of Raspbian:**

```
git clone https://github.com/DexterInd/GrovePi

curl -kL dexterindustries.com/update_grovepi | bash

cd /home/pi/Dexter/GrovePi/Firmware
bash firmware_update.sh
```

**But now I'm getting the following error:**
```
pi@raspberrypi:~/Dexter/GrovePi/Firmware $ bash firmware_update.sh
Updating the GrovePi firmware
=============================
 http://www.dexterindustries.com/grovepi 
 Run this program: 
 sudo ./firmware_update.sh
 
=============================
Do you want to update the firmware? [y,n]y
Make sure that GrovePi is connected to Raspberry Pi
Firmware found
Press any key to start firmware update
. . .davrdude: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory
avrdude: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory
avrdude: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory
avrdude: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory
```
**Raspberry Pi Specs:**
Raspberry Pi 3B+
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

erikvandorp, 2020-03-22:
I have the same problem.

CleoQc, 2020-04-01:
Run `sudo apt-get install libncurses5`
It seems this library seems it's now missing from Raspbian..

closed 2020-04-01 by CleoQc