DexterInd/GrovePi #478: Home Weather Station -- Runtime Error

jonathanrjpereira, 2020-03-13:
**Installation Steps:**
```
git clone https://github.com/DexterInd/GrovePi
curl -kL dexterindustries.com/update_grovepi | bash
```

**Error:**
```
Python 3.7.3 (/usr/bin/python3)
>>> %cd /home/pi/Dexter/GrovePi/Projects/Home_Weather_Display
>>> %Run Home_Weather_Display.py
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/smbus_cffi-0.5.1-py3.7-linux-armv7l.egg/smbus/smbus.py", line 81, in open
    self._fd = os.open(path, os.O_RDWR, 0)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/i2c-1'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Dexter/GrovePi/Projects/Home_Weather_Display/Home_Weather_Display.py", line 41, in <module>
    from grove_rgb_lcd import *
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/usr/local/lib/python3.7/dist-packages/grovepi-1.0.4-py3.7.egg/grove_rgb_lcd.py", line 33, in <module>
    bus = smbus.SMBus(1)
  File "/usr/local/lib/python3.7/dist-packages/smbus_cffi-0.5.1-py3.7-linux-armv7l.egg/smbus/smbus.py", line 56, in __init__
    self.open(bus)
  File "/usr/local/lib/python3.7/dist-packages/smbus_cffi-0.5.1-py3.7-linux-armv7l.egg/smbus/smbus.py", line 83, in open
    raise IOError(e.errno)
OSError: 2
```

**Raspberry Pi Specs:**
Raspberry Pi 3B+
Distributor ID: Raspbian
Description: Raspbian GNU/Linux 10 (buster)
Release: 10
Codename: buster

DAnGioi, 2020-03-29:
You can try this: curl -kL dexterindustries.com/update_grovepi | bash -s -- --user-local

DAnGioi, 2020-03-29:
Help me！My GrovePi’s RST often light up after hours of work .My GrovePi’s RST often light up after hours of work.

DAnGioi, 2020-03-29:
Are you having the same problem?My GrovePi’s RST often light up after hours of work

jonathanrjpereira, 2020-04-01:
I'm not having problems with any of the GrovePi lights.

CleoQc, 2020-04-01:
@jonathanrjpereira 
what do you get if you run `sudo i2cdetect -y 1` ?
It seems i2c isn't enabled.

CleoQc, 2020-05-19:
Closing since no answer

closed 2020-05-19 by CleoQc